qmk/qmk_firmware #293: Key codes for KC_MNXT and KC_MPRV

romanzolotarev, 2016-04-28:
Looks like `KC_MNXT` and `KC_MPRV` send key codes different from code sent by MacBook built-in keyboard.

### ErgoDox

```
code:0x11       misc:ConsumerKeyCode::MUSIC_NEXT_17
code:0x12       misc:ConsumerKeyCode::MUSIC_PREV_18
```

### MacBook Keyboard

```
code:0x13       misc:ConsumerKeyCode::MUSIC_NEXT
code:0x14       misc:ConsumerKeyCode::MUSIC_PREV
```

Is there a way to change codes?

P.S. Here is my layout https://github.com/jackhumbert/qmk_firmware/pull/258

eltang, 2016-04-28:
See [this documentation](https://github.com/tmk/tmk_keyboard/wiki/FAQ-Keymap#media-control-keys-in-mac-osx)!

romanzolotarev, 2016-04-28:
@eltang wow, thanks!

closed 2016-04-28 by romanzolotarev

romanzolotarev, 2016-04-28:
Tried `KC_MFFD` and `KC_MRWD` keys codes. Does not work for me. :(