qmk/qmk_firmware #296: MOUSEKEY_DELAY should behave differently with mouse movement and mouse scroll

BeatVids, 2016-04-30:
The current default for the mouse delay is:
`#define MOUSEKEY_DELAY          100`

I actually did not like that delay at all for mouse movement, so I changed it to:
`#define MOUSEKEY_DELAY          0`

However, today I decided to finally add the mouse scroll buttons to my layout and I realized that the delay is pretty useful for the mouse wheel scroll buttons. With the delay being 0, the mouse wheel scroll buttons lose precision, and the scrolling is a bit too fast, even if quickly tapping it.

Therefore, I was thinking a new variable would be really useful.
This would be my ideal config, and I feel like many of you would agree:
`#define MOUSEKEY_DELAY          0`
`#define MOUSEKEY_SCROLL_DELAY          100`

ezuk, 2016-05-03:
This is actually something you might be able to implement. Did you look around to see where MOUSEKEY_DELAY is used? By strategically changing some of these to use your new `MOUSEKEY_SCROLL_DELAY`, you might be able to get what you want. It doesn't have to be perfect -- give it a try, open a pull request, and @jackhumbert will probably review it and help :)

BeatVids, 2016-05-03:
Hi Erez!
Sure, I'll give it a shot!
After a quick grep search, I think it's going to be in one of these 3

```
qmk_firmware-master/keyboard/ergodox_ez/ergodox_ez.lss
qmk_firmware-master/keyboard/ergodox_ez/obj_ergodox_ez/common/mousekey.lst
qmk_firmware-master/keyboard/ergodox_ez/obj_ergodox_ez/common/command.lst
```

For a head start, do you know which one will be it?

eltang, 2016-05-03:
`mousekey.c` is where you want to look!

jackhumbert, 2016-05-03:
Those are the compiled object files - the source file is [here](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/common/mousekey.c#L42) :)

eltang, 2016-05-03:
@jackhumbert Those are actually assembly listings!

ezuk, 2016-05-03:
Hey @eltang -- allow me to jump in here with some quick etiquette/"how to help" feedback. We eventually might create a code of conduct, but for now I'd like to chat here. So:

1) Thank you for participating and for wanting to help! That's wonderful and highly encouraged. What I write below isn't to discourage you, but to help you help others better.

2) Speed is nice when replying - but the level of detail is more important. So take a few moments and be as specific as you can -- if you're able to point to a specific line in a file like Jack did, that's wonderful. If you can even offer some code like you sometimes do (after you test it!) that's even better. So time is less important than quality.

3) Try to keep threads on point. "Those are actually assembly listings!" -- this comment has no bearing on the current discussion, and I would actually delete it if I weren't referring to it right now.

4) "The exclamation mark (Commonwealth English) or exclamation point (American English) is a punctuation mark usually used after an interjection or exclamation to indicate *strong feelings* or *high volume (shouting)*, and often marks the end of a sentence." (From [Wikipedia](https://en.wikipedia.org/wiki/Exclamation_mark))

BeatVids, 2016-05-03:
Hi guys, after staring at this for a while, I think I know what to do, I just don't know how to do it.

In [Line 99](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/common/mousekey.c#L99), I'm thinking of adding 2 lines for mk_wheel_delay.

What must I do to only run the first part when a mouse move is being pressed?
What must I do to only run the second part when a mouse scroll is being pressed?
```
    if (timer_elapsed(last_timer) < (mousekey_repeat ? mk_interval : mk_delay*10))
        return;

    if (timer_elapsed(last_timer) < (mousekey_repeat ? mk_interval : mk_wheel_delay*10))
        return;
```

eltang, 2016-05-03:
@BeatVids Which delay do you want to use when moving and scrolling are performed at the same time?

BeatVids, 2016-05-03:
Great question, although that's pretty much never going to happen intentionally, I would rather use the mk_delay (Set to 0)

eltang, 2016-05-03:
@BeatVids I've got a pretty simple implementation right now. I'll have to test it when I have time

BeatVids, 2016-05-03:
Thank you Eric!
I would love to see what you have so far, maybe I can play with it also.

eltang, 2016-05-03:
I didn't need to change more than a few lines of code. It would be great if you could test  as I am unfamiliar with your problem. My complete `mousekey.c` file can be found [here](https://gist.github.com/eltang/f8d50067e95e2820241ec74abf3f94a8).

BeatVids, 2016-05-03:
@eltang That's perfect, thank you!
I chose to replace:
`MOUSEKEY_MOVE_DELAY` and `mk_move_delay`
with their original names:
`MOUSEKEY_DELAY` and `mk_delay`
Since that variable is used in other places, it was giving me issues.

@jackhumbert I hope this can added to the default mousekey.c sometime.
Moving the mouse no longer has that unnecessary delay, and moving the mouse wheel has a useful delay that allows me to tap it in a relaxed manner, to let me scroll only one unit. I don't have to tap it instantly anymore, without getting multiple scroll units in one tap.
This is how my config now looks like:

```
#define MOUSEKEY_DELAY          0
#define MOUSEKEY_WHEEL_DELAY    200
```

Thanks again @eltang

eltang, 2016-05-04:
@jackhumbert Should I open a PR (when you're less busy)?

eltang, 2016-05-04:
@BeatVids Everything works as expected when you try to move and scroll at the same time, yes?

BeatVids, 2016-05-04:
Yes, just like you asked initially!
If the mouse is already moving, there is no delay for the mouse wheel, which is perfectly fine.
I honestly don't know if anyone would prefer it with the delay, as I can't see anyone that actually intends on using both at once like that frequently.

eltang, 2016-05-04:
Awesome! Sorry for the compilation errors—I honesty didn't think those variables were used anywhere else!

closed 2016-05-04 by BeatVids

ezuk, 2016-05-04:
So great, @eltang!! :)

eltang, 2016-05-04:
@BeatVids I was actually asking what happens when you try to *start* moving and scrolling and moving at the same time, not one other.

BeatVids, 2016-05-04:
The answer is the same, they do not conflict at all with each other.

I tried to mix with mouse move + mouse scroll, and mouse move + mouse move
Both simultaneously and almost simultaneously (right after another)

I looked at the issue you referenced, and I did not experience anything like that whatsoever.

Maybe I'll notice some other random problem in the near future, but it's flawless for the way I've been using it thus far.

eltang, 2016-05-04:
It seems that I'll have to separate the code for moving the wheel and moving the mouse a lot more for two delays to work.