qmk/qmk_firmware #299: Adding a Colemak layer in addition to the default firmware.

Ormia, 2016-05-02:
Hi,

I would like to retain the original 3 default layers that come with the ergodox ez but also add an additional colemak layout layer. I can't seem to achieve this with the massdrop configurator because I cannot program the Media and Mouse layer. Can you help me achieve this setup?

Thanks

eltang, 2016-05-02:
QMK will allow you to edit [the default layout](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboard/ergodox_ez/keymaps/default/keymap.c) to your heart's content! Check out [these instructions](https://github.com/jackhumbert/qmk_firmware#getting-started) to get started!

ezuk, 2016-05-03:
@Ormia just looping back to ask if you're sorted?

Ormia, 2016-05-03:
Hi Ezuk, I haven't been able to add the additional Colemak layer yet because I am not familiar with writing firmware and 'c' code. It seems to be a steep learning curve for me at the moment.

ezuk, 2016-05-03:
@Ormia understood. Just out of interest, why are you not using a software-defined Colemak implementation? I'm asking because that's what I do, even though I know how to program the ErgoDox EZ -- I've been using Colemak on Windows and Mac for many years now, and I find the software implementations very good. So my keyboard sends QWERTY but I type Colemak.

Is there a reason you don't want to go this route?

Ormia, 2016-05-03:
@Ormia <https://github.com/Ormia>  understood. Just out of interest, why are you not using a software-defined Colemak implementation? I'm asking because that's what I do, even though I know how to program the ErgoDox EZ -- I've been using Colemak on Windows and Mac for many years now, and I find the software implementations very good. So my keyboard sends QWERTY but I type Colemak.

Is there a reason you don't want to go this route?

 

I have never even considered using software implemented Colemak as an option. Can you recommend software that will do this for me?

Thanks

ezuk, 2016-05-03:
@Ormia for sure! http://colemak.com/Download - this is all you need :) Let me know if that works.

BeatVids, 2016-05-04:
@ezuk , I am quite surprised you use a software-implemented Colemak. May you please let my know why? The fact that there are some places that will not grant me administrator rights to change the layout was the sole reason I looked up alternative keyboards. I was pissed at first, but I'm glad now since I have the best keyboard in the world. I recommend trying out the Colemak-DH mod, I've been using it since I got my EZ, it's better on the index fingers! You are the coolest CEO ever!

@Ormia , what OS are you using? If you are using Linux by any chance, I will be happy to help to help you install everything and to get started. I am just like you, don't know squat of C, but I managed to get it working perfectly (and more!). It's pretty EZ actually, the hardest part was installing everything, but now changing the layout is a piece of cake. I truly recommend doing it this way, because there is only so much you can do with a software implemented setup. Like using the rarely used letter "Z" as a ctrl if you hold it, spoiling your pinky a bit.
I was also able to get 2 new features, I definitely didn't do it on my own since like I mentioned I don't know C, but thanks to this awesome little community including @eltang, I was able to do them.
One of them is tapping one key to paste (ctrl+v), holding it slightly to copy (ctrl+c), and holding it for half a second to cut (ctrl+x).
And something else that may motivate you, if you wanted to use your keyboard on a friends computer, you have to always toggle the colemak on and off. And some places won't even let you do this.

ezuk, 2016-05-05:
Hey @BeatVids! :) First, thank you for your kind words!! :)

There are a few reasons why I use software-implemented Colemak:

1) In addition to my desktop workstation (ErgoDox EZ), I also use two laptops every day. I just need Colemak there, so that's part of it.
2) I was using software Colemak before I made the EZ, and it's still working well, so hey. :)
3) I'm bilingual -- and Hebrew actually uses the QWERTY mappings (I mean, has its own layout obviously). If I hardware-map to Colemak, it'll mess up the Hebrew layout (unless I switch layers of course, so I still need to maintain a QWERTY layer on my keyboard)
4) Colemak-DH sounds interesting! You are not the first to recommend it, I might look into it. One thing that's keeping me in "traditional" Colemak is that it's available out of the box in OSX and Linux.

BTW, I love your copy/cut/paste key idea. Did you see I featured it in the [main repo readme](https://github.com/jackhumbert/qmk_firmware#example-1-single-key-copypaste-hold-to-copy-tap-to-paste)? Did you ever make a pull request with your layout?

Your reasons for using the hardware mapping make total sense -- I'm very happy the EZ enables you to do this :)

Also, hat-tip to @eltang for all of the help! That's awesome, Eric! :)

Ormia, 2016-05-05:
@ ezuk. Thanks for the recommendation. I'll give this a try first and at some point when i have more free time, I'll explore coding the ergodox.

@BeatVids. i'm using Win7. Thanks for you offer in helping me.

ezuk, 2016-05-05:
@Ormia - Are you able to follow the instructions in [this link](https://github.com/jackhumbert/qmk_firmware/blob/master/BUILD_GUIDE.md#windows) to set up the build environment on Windows? Do you need a hand with these steps?

Ormia, 2016-05-06:
I think I will be ok from here. Thanks!

ezuk, 2016-05-06:
Yay! Happy to hear it, and good luck! :)

BeatVids, 2016-05-07:
Hi @ezuk , I promise I'll submit my layout sometime. I'm just taking my time to create my most perfect layout. It's like a really hard rubiks cube to put everything perfectly :D. I'm aiming to make a nice image too just like yours, so people can have an easier visualization preview. I'll also create a default_colemak_dh keymap!

I did a pull request to add in the even longer key hold for the cut (ctrl+x) feature.
I'm very happy my idea made a difference :)
Thanks Erez

ezuk, 2016-05-07:
Awesome! My advise here: Don't wait for it to be perfect. Look at what @romanzolotarev and @nrrkeene did with their keymaps. They submitted something and kept improving it over time, submitting multiple pull requests, until it got to a super high level of polish. Just a thought :)

romanzolotarev, 2016-05-07:
@ezuk thank you for mentioning me. @BeatVids here is an example: https://github.com/jackhumbert/qmk_firmware/pull/258

## Recipe 

- use http://www.keyboard-layout-editor.com (it saves your layout to https://gist.github.com)

- make a screenshot

  For example:

  - http://www.keyboard-layout-editor.com/#/gists/8ebcb701ecb763944417
  - http://www.keyboard-layout-editor.com/#/gists/9e89d54f1ea6eeeb7dab1b2d19d28195

- edit your `keymap.c` (e.g. [mine](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboard/ergodox_ez/keymaps/romanzolotarev-norman-qwerty-osx/keymap.c))

- test it and make PR

- attach screenshot

- and link to http://www.keyboard-layout-editor.com to help others to fork it :)

closed 2016-06-18 by jackhumbert