qmk/qmk_firmware #301: ergodox ez: ⌃⌥⌘-power works with slightly edited default, but not my layout

adiabatic, 2016-05-03:
I went and added a power button (KC_PWR) to [my layout](https://github.com/adiabatic/qmk_firmware/tree/ergodox-zweihander-osx/keyboard/ergodox_ez/keymaps/zweihander-osx), Zweihander.

Pressing and holding ;/media (layer 2) and Power will reliably bring up the usual "are you sure you want to restart/sleep/shut down/cancel?" dialog, but none of the other power-related keyboard combos I've tried do anything reliably, including:

- ⌃⌘-power (restart)
- ⇧⌃-power (sleep displays)
- ⌃⌥⌘-power (shut down)

However, I remember that ⌃⌥⌘-power worked on an earlier version of my keyboard layout, so I went back to the default keymap, added a Power button to the media layer, reflashed my keyboard, and all the above keyboard shortcuts worked just fine. Any idea why, and what I can do to have both my customized layout and fast restarts/shutdowns/display sleeping? Teensy says "ergodox_ez.hex, 57% used" when I flash it, so I don't _think_ I'm sending it too-large .hex files…

my usual layout: [zweihander-osx.zip](https://github.com/jackhumbert/qmk_firmware/files/246264/zweihander-osx.zip)

layout with a _fully_ working power button: [default-and-power.zip](https://github.com/jackhumbert/qmk_firmware/files/246263/default-and-power.zip)

eltang, 2016-05-03:
What happens when you use `KC_F12`?

jackhumbert, 2016-05-03:
Could you start up `xev` and see what keys are actually being sent in both firmwares? That might help sort out the differences.

adiabatic, 2016-05-03:
@jackhumbert I don't have any machines here running the X Window system on bare metal, and `xev` doesn't seem to be in Homebrew.

skullydazed, 2016-05-03:
@adiabatic It comes with XQuartz. Just launch X11.app (either via spotlight or by going to /Applications/Utilities) and if you don't have it installed already you will be prompted to download and install it.

adiabatic, 2016-05-03:
@eltang I started up Keyboard Viewer in OS X and while the rest of the function keys seem normal, F12 often fails to light up when I press it. When I press and unpress and press and unpress it, it'll briefly flicker (measure the time it's on in milliseconds). This would explain why it's so flaky…

Tried holding ⌃⌥⌘ and pressing the F12 key repeatedly (on layer 1, with layer 1 lock on), but it took a bunch of keypresses of F12 to get the shutdown sequence to start.

adiabatic, 2016-05-04:
@jackhumbert [here's my xev session](https://www.frogorbits.com/Images/xev.mov). I press ⌃⌥⌘-power and ⌃⇧-power (sleep). Everything seems to work fine for me here _right now_.

Several revisions of my keyboard ago I changed two keys on the media layer from KC_TRNS to something, reflashed, and pretty much only the right ergodox half worked (a couple of keys on the left worked). I stared at the changes I made, concluded I didn't do anything wrong, hit Undo a bunch of times, saved, rebuilt, reflashed, and everything worked again. Then…I redid those changes (⇧⌘Z), saved, rebuilt, reflashed, and everything worked with the two new keys.

Apparently it's better behaved for this boot-up (I reboot into Windows fairly frequently, plus I shut this machine off at the end of the night). I'll see if I can get it to behave differently later this evening or tomorrow.

adiabatic, 2016-05-30:
Here's another `xev` log, with description of what I did: <https://gist.github.com/adiabatic/1c813dda5489409648d3f5bc8a9c5637>.

After shutting down my machine and turning it back on again — without reflashing anything in the meantime — I'm trying the keyboard combination and now my machine is falling asleep (⌥⌘-power?) instead of shutting off (⌃⌥⌘-power).

Now I strongly suspect that the title of this bug report is wrong.

jackhumbert, 2016-05-31:
That's pretty bizarre. My only guess is that it would be something with the EEPROM (on the firmware side), which can be reset with space+backspace while plugging in the keyboard, or the [eeprom reset tool](https://github.com/jackhumbert/qmk_firmware/blob/master/quantum/tools/). Otherwise, my guess is that it'd be something on the OS side of things.

It could be something with an uncleaned workspace as well - running `make clean` in-between builds will prevent that.

adiabatic, 2016-05-31:
Weird, I haven't had dfu-programmer. Installed now. `make clean && make KEYMAP=zweihander-osx` hadn't been outright failing even though it wasn't being found in my $PATH, so I didn't worry about it. Also unplugged my keyboard held space and backspace, and plugged it back in.

Not sure how I managed to install two out of three things from Homebrew.

ezuk, 2016-06-01:
That's pretty weird, that this is how the issue manifested. Did installing dfu-programmer and `make clean` did fix it? Can close out?

adiabatic, 2016-06-01:
I tried to reboot into Windows and holding the alt/option key didn't bring up the pick-a-partition prompt during boot. The button seems to work fine in OS X, though, and it worked during boot yesterday…

At any rate, dfu-programmer doesn't seem to notice my keyboard:

```shell
$ dfu-programmer atmega32u4 erase # from the readme in quantum/tools
dfu-programmer: no device present.
```

I have no idea if there's an atmega32u4 in my keyboard; that could be why my manual reset isn't doing anything.

Other, possibly related weirdness: While I was trying to reboot into Windows and getting dumped into OS X, I tried to type with the keyboard and no letters showed up. I unplugged the keyboard, plugged it back in, and everything seemed to work OK.

I suppose the thing to try next is to try the dfu-programmer-based erase-and-reset and then reflash my current layout with Teensy, right?

ezuk, 2016-06-01:
Yeah, basically dfu-programmer isn't really supposed to work with the Teensy, which is why I find this whole thing a little confusing. I'd be curious to see what erasing with it does, yeah.

closed 2016-07-24 by ezuk