qmk/qmk_firmware #306: KC_MENU produces Sun Props key

Gamayun, 2016-05-04:
It appears KC_MENU results in my ErgoDox EZ sending keycode 138 / SunProps rather than 135 / Menu.
Is this intentional or..? 
And if so, how should I go about getting a working Menu key?
The menu key might not be that popular, but I can't think of many use cases for the Props key... ;-)

`KeyPress event, serial 40, synthetic NO, window 0xa00001,
    root 0xd8, subw 0x0, time 33601535, (150,-63), root:(150,663),
    state 0x10, keycode 138 (keysym 0x1005ff70, SunProps), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False
`
versus
`KeyPress event, serial 40, synthetic NO, window 0xa00001,
    root 0xd8, subw 0x0, time 33610194, (150,-63), root:(150,663),
    state 0x10, keycode 135 (keysym 0xff67, Menu), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False
`

jackhumbert, 2016-05-05:
[Looking at this](https://input.club/forums/topic/ergodox-menu-key-not-working), it appears you need to use `KC_APP` instead? I'm not sure where the name mix-up is, but my guess is it's some sort of PC vs Unix argument :)

Gamayun, 2016-05-07:
Ah, right! Thanks!
Yep, that appears to be the case. Both on PC and *nix.
Looks like it is the 'correct' naming - going by the wisdom of the USB HID specs.
Might be an idea to note it somewhere in the docs though. :-)

closed 2016-05-07 by Gamayun