qmk/qmk_firmware #318: KC_LGUI/KC_RGUI randomly stopped working

jdavis, 2016-05-11:
I sent an email to @ezuk asking about it and he mentioned to open an issue. Whether or not this is hardware issues or software, I'm not sure.

It doesn't matter what keys I bind KC_LGUI/KC_RGUI to, none of them ever trigger on my computer. I know it isn't computer specific, I've tried it on multiple computers. It only appears to be L/R GUI; all other keys work fine.

I have two ErgodoxEZ's but only one is having this problem yet they both use the same layout/firmware.

My layout is here although I don't think it is the problem: https://github.com/jdavis/ergodox

Any ideas?

eltang, 2016-05-11:
It sounds like you accidentally configured Boot Magic. You can clear the settings by holding your Space and Backspace keys as you plug in your keyboard. After that, you can recompile with `BOOTMAGIC_ENABLE = no` if you wish for this not to happen again.

jdavis, 2016-05-11:
That's fascinating. How could I have accidentally configured this during use?

eltang, 2016-05-11:
What's most likely is that you were holding down those keys and your spacebar one of the times you were plugging in your keyboard.

jdavis, 2016-05-11:
This worked! Somehow those keys must have been pressed while plugging it in.

Thanks for the pointers, @eltang.

closed 2016-05-11 by jdavis