qmk/qmk_firmware #319: [experimental keymap] Raise and Lower keys cause keyboard freeze.

Bilger321, 2016-05-11:
- Keycodes are being sent and backlight cycles when bottom left button is pressed, but on pressing the RAISE or LOWER keys, keyboard freezes. 
- No more keycodes are sent, backlight no longer cycles, and Windows makes the "Disconnect" and "Reconnect" sounds. 
- When LEDs are completely off, LEDs glow faintly after keyboard failure. 
- Green LED on PCB is still lit after failure.
- Unplugging and replugging the keyboard resets it, but pressing the RAISE or LOWER keys still cause failure. 
- Hardware is REV 3 PCB with no modifications. (No speaker, LEDs are installed.) Software is Windows 10, Keyboard programmed using ATMEL Flip without prior issues.

jackhumbert, 2016-05-11:
Were you the one on reddit with the same issue solved by [the eeprom reset tool](https://github.com/jackhumbert/qmk_firmware/tree/master/quantum/tools)? 

At any rate, thanks for sticking it here :)

Bilger321, 2016-05-11:
Yes that was me! The issue can be solved without the use of the EEPROM reset tool as well, as far as I can tell most of the issues other the ones listed here were due to my error with using ATMEL Flip. All that has to be done after keyboard failure is just loading up a different HEX file.

jackhumbert, 2016-05-11:
Ah, gotcha. Thanks!

jackhumbert, 2016-05-18:
Hey - just wanted to double check with this - did you ever try recompiling and reflashing the same layout? I wasn't able to reproduce it on my end, and I have similar hardware (rev 3).

closed 2016-06-18 by jackhumbert