qmk/qmk_firmware #322: Enable/Disable Plover when toggling the Plover Layer

cwhits, 2016-05-12:
There's a keyboard shortcut in plover for toggling whether output is enabled or disabled. Would it make sense to add this? Is it universal between Mac/Windows/Linux?

jackhumbert, 2016-05-12:
Sure! There's a couple of free spaces on the Planck Plover layer - do you know what it is in the default Plover setup?

cwhits, 2016-05-12:
So, the only issue with that pull request is you'll see `erfvol` if plover isn't open. It's convenient if it _is_ open that you don't have to do any extra keystrokes and makes it function the same as changing any other layer. I could see the benefit of having the toggle be a completely separate key though.

Let me know your thoughts.

jackhumbert, 2016-05-12:
Ah, I thought it might have been tied to an actual key on the keyboard. Let's hold off on that, then - I'd like for people to be able to switch to the Plover layer without a bunch of things printing out.

eltang, 2016-05-12:
@jackhumbert Have the default keymaps been finalized yet?

PR #323 (2016-05-12): [planck] toggle plover output in app when toggling plover layer on kb
The default plover command for toggling output on and off is `"PHROLG": "{PLOVER:TOGGLE}"` (at least on OS X).

This commit sends those characters when the layer is toggled on and off on the Planck.

As long as Plover is running, no output is shown. If Plover is _not running_ `erfvol` will be output when toggling the Plover layer is toggled on and off.

Fixes #322
closed 2016-05-12 by jackhumbert