qmk/qmk_firmware #344: ONESHOT_TIMEOUT not timing out - firing off character repeatedly

chrischambers, 2016-05-23:
The behaviour is working as expected in the 'townk_osx' keymap, for reasons I can't identify. I've tested in my own keymap as well as in minimalist versions based on some of the existing keymaps.

To reproduce, take any keymap with >= 2 layers,  set a key to `OSL(1)` in the base layer, and set `ONESHOT_TIMEOUT = 3000`, say, in your `config_user.h`

**Expected behaviour:** 

Given the above, I should be able to press the OSL(1) key, wait 3 seconds, and have the layer revert back to the base layer. Alternatively, I should be able to hit the OSL(1) key and then hit another key to have its equivalent layer 1 mapping fire.

**Actual behaviour:**
If you press X and wait more than the 3 seconds, the layer doesn't revert back to base: when you then hit a key, it will repeatedly fire off the layer 1 mapping for that key until you touch the keyboard again. The alternate behaviour does work as expected.

chrischambers, 2016-05-25:
@Townk - any idea why `ONESHOT_TIMEOUT` is working properly in your keymap but isn't elsewhere?

Townk, 2016-05-25:
Sorry for the late response. I'll check it out tonight to see if I can find something.

chrischambers, 2016-05-25:
That'd be awesome, thanks. :)

Townk, 2016-05-26:
Chris, do you have your key map in some place so I can check it out?

chrischambers, 2016-05-26:
Sure, it's available [here](https://github.com/chrischambers/qmk_firmware/tree/chrischambers/keyboard/ergodox_ez/keymaps/chrischambers). Having said that, I don't think it's particular to my mapping - I believe you can take any of the ergodox_ez mappings, make the alterations I've listed above and encounter the same issues. I think I tested with the default mapping and the kines-ish one.

Townk, 2016-05-26:
Try to use the `fn3` key for the one shot layer with time out. That is the difference between your key map and mine. Also it explains why I was able to change the default and have it working :).

Townk, 2016-05-26:
Btw, if that works we know the problem is on the OSL macro.

chrischambers, 2016-05-26:
Pretty sure I've already tried that, but I'll test again now...

chrischambers, 2016-05-26:
Yup, it's not a problem with the `OSL` macro, same issue occurs when I use `ACTION_LAYER_ONESHOT`.

Townk, 2016-05-26:
Well, you're right! I changed my layout to use the macros instead and it's still working. Also I realized that I didn't test the default keyboard yesterday and indeed the OSL does not behaves as expected. I'm trying to get my layout to be like someone else's to see what is different and I'll get back to you once I have something.

Townk, 2016-05-26:
ok, I could even reproduce the problem on my layout. If I try to use OSL on my KEYPAD layer I have the same behavior as you. I tried to change the layer values of default keyboard (from 1 which is my KEYPAD to 2 which is my FN) but the problem persist.

At this point I'm pretty sure is something on the action.c file. I'll try to setup a console here to try to debug it but I'm not finding any documentation on how to do so. If anyone have any ideas on the original problem or how to get a console here, let me know.

ezuk, 2016-05-28:
@jackhumbert thoughts?

chrischambers, 2016-06-03:
Any progress @Townk ?

Townk, 2016-06-06:
Didn't have time to dig into this yet, but is in my todo list.

ahtn, 2016-06-18:
Could you please provide a keymap which has this bug? I tried to reproduce it on my branch (not QMK) and I couldn't reproduce it.

chrischambers, 2016-06-18:
@ahtn : I've linked to mine in the 6th post in this thread, but as said, it applies to any of the ergodox_ez keymaps (steps to reproduce in the first post).

ahtn, 2016-06-18:
Ah thanks, how did I miss that? Tried again on my branch and I still couldn't reproduce this, so it's probably specific to the ergodox ez or QMK.

@Townk To get console output for debugging, you use the [HID listen program](https://www.pjrc.com/teensy/hid_listen.html). Also, add these flags to your makefile

    EXTRAFLAGS += -DDEBUG -DDEBUG_ACTION

See [debug.h](https://github.com/jackhumbert/qmk_firmware/blob/b79400e0e9080112fbdcf9ae1672297286b7938a/tmk_core/common/debug.h) and [print.h](https://github.com/jackhumbert/qmk_firmware/blob/b79400e0e9080112fbdcf9ae1672297286b7938a/tmk_core/common/print.h) for more info.

ezuk, 2016-06-21:
So @jackhumbert and I took quite a bit of time tonight to dig into this. Long story short, definitely able to replicate on my end -- this is a confirmed bug.

What's interesting is that @Townk's keymap relies on code that he added to action.c, but it only works with his keymap. The problem here is that this is nestled deep within `action.c`, which is very hard to decipher. We're going to take much of the stuff implemented in this file and essentially rewrite it into `quantum.c` in a way that we feel is easier to understand, and then we'll also be able to fix this issue.

I know this is a small comfort, but let's keep this issue open so this stays on the radar. And @Townk if you feel like digging in, that would be amazing :)

Townk, 2016-06-22:
I want to give it a try but no matter what I do I can't get the HID listen program to show anything. When I add -DDEBUG to EXTRAFLAGS I get this error:

`Compiling: matrix.c                                                   [ERRORS]
 |
 | <command-line>:0:12: error: expected ',' or '}' before '-' token
 | ../../quantum/keymap.h:88:5: note: in expansion of macro 'DEBUG'
 |      DEBUG,
 |      ^
 |
make[1]: *** [.build/obj_ergodox_ez_townk_osx/matrix.o] Error 1
make: *** [all] Error 2`

ahtn, 2016-06-22:
Ops, looks like you also need this in the makefile

    CONSOLE_ENABLE = yes # Console for debug(+400)

Even if the `-DDebug` flag is still casing errors, you should still be able to use the functions from [print.c](https://github.com/jackhumbert/qmk_firmware/blob/b79400e0e9080112fbdcf9ae1672297286b7938a/tmk_core/common/print.h) with `CONSOLE_ENABLE`.

I haven't used QMK, so it's possible something might have changed from TMK.

Townk, 2016-06-24:
I found some flaws on the one shot layer implementation and now I'm working to find a fix for it. I still didn't find the reason why it's working on my keymap but I found that even on my keymap there are bugs (the one shot layer does not time out as expected). I'll update you more after I spend more time with this code over the weekend.

jackhumbert, 2016-06-25:
@Townk are you interested at all in moving the implementation into the new quantum format? I'm working on making that more standardised [here](https://github.com/jackhumbert/qmk_firmware/tree/quantum-keypress-process/quantum/process_keycode), where each feature's function will get called from the `process_record_quantum` function in `quantum.c`. I'm hoping eventually all of the shortcuts we have setup (MT, MO, LT, OSL, etc) will be implemented here in their own files instead of in `action.c`.

chrischambers, 2016-07-19:
Status update? :)

Townk, 2016-07-21:
I probably will look into the new way once is available but right now I don't have time to investigate (family is getting bigger and we started to do classes after my work hours :P). One thing though, I try to check the link @jackhumbert pasted here (https://github.com/jackhumbert/qmk_firmware/tree/quantum-keypress-process/quantum/process_keycode) but I get a 404 there. Is there a branch that I should checkout for it?

algernon, 2016-07-21:
It's been merged to [master](https://github.com/jackhumbert/qmk_firmware/tree/master/quantum/process_keycode), you'll find the directory there.

jhenahan, 2016-07-23:
For what it's worth, I got around this issue by skipping `OSL` altogether and doing this:

https://github.com/jhenahan/qmk_firmware/blob/0bdf9ab7c7a6e1c48f0500bba0d95206de47dd46/keyboards/planck/keymaps/jhenahan/keymap.c#L201

chrischambers, 2016-12-27:
Any further updates?

rclasen, 2017-04-02:
this appears to be fixed with 6f44ca7a59 (and likely some further commits)

skullydazed, 2017-06-27:
Closing based on @rclasen's comment. @chrischambers if you find this is still broken please reopen.

closed 2017-06-27 by skullydazed