qmk/qmk_firmware #347: Support for PS2_MOUSE_ENABLE flag from TMK?

tyetye, 2016-05-24:
I am trying to enable support for this flag from the TMK firmware. The guts are in QMK but I haven't been able to get it working so far. @jackhumbert commented that this might be in the works a few months ago: [https://www.reddit.com/r/olkb/comments/3w1eid/jacks_weekly_ama_for_tuesday_20151215_8pm_est_all/cxyieoi] .
Before I spend too much time on this:
1. Jack, do you still hope to implement this?
2. I know QMK has veered pretty far from TMK. So, are you aware of any reason that external mouse support/PS2_MOUSE_SUPPORT shouldn't work anymore?

jackhumbert, 2016-05-24:
1. I do, but I probably won't have time to mess around with it for another couple weeks/months.
2. As long as you have that option in the Makefile turned on, it should!

jackhumbert, 2016-09-13:
I would like to package this as it's own feature (as a Makefile option) with a brief schematic/explanation for people looking to implement it :)

jackhumbert, 2017-06-27:
I'm not sure when it changed, but I think this is supported now, [as several new keyboards use it](https://github.com/qmk/qmk_firmware/search?utf8=%E2%9C%93&q=PS2_MOUSE_ENABLE&type=) - if I'm mistaken, @tyetye, please reopen this issue!

closed 2017-06-27 by jackhumbert