qmk/qmk_firmware #350: Numpad Keycode numbers and keycode list

squarology, 2016-05-25:
Hi, what keycode should I use to create numpad (as opposed to regular numbers) numbers? Is there a guide somewhere? I tried looking but I couldn't find anything.

There are many keys and a lot of them are obvious but still I'd like to know if there a guide somewhere explaining some of the lesser known keys like button, mice, function, miscellaneous, lang keys, etc.

skullydazed, 2016-05-25:
The keypad codes are all prefixed with KP, so you'll use KC_KP1, KC_KP2, etc.

There's an overview of the keymap definitions here: https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/doc/keymap.md

And you can find every single key code documented here: https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/doc/keycode.txt

squarology, 2016-05-25:
@skullydazed wow, thanks a lot. just what I was looking for.

closed 2016-05-25 by squarology