qmk/qmk_firmware #355: [Enhancement] Leader key tap/hold dual function.

BeatVids, 2016-05-28:
Hi gentlemen!
Just came across the documentation for a feature I've anticipated for a while now, thanks so much for your hard work into making it :)

Since the leader key is basically just a "tap" key, I was anticipating on making that key have a dual purpose for holding it.

As the new doc says, "it's a single action key, can't be used for anything else."
So, my idea doesn't work, but let's just say the following hypothetically works:

Let's say I have a macro layer (MCRO) that has various macros for it, just how we had macros before the leader key.

Let's call this the MacroKey, and set it to a layer with macros, and the leader key: LT(MCRO,KC_LEAD)

```
HOLD MacroKey + e       = email@address.com
TAP  MacroKey , e       = ErgoDox
TAP  MacroKey , e , z   = ErgoDox EZ
```

That button doesn't have to be all about macros, it could also be used for the MDIA layer for example.

```
HOLD MacroKey + p       = KC_MEDIA_PREV_TRACK
TAP  MacroKey , p       = PasswordDon'tHackMePlz
```

I completely understand if it's asking for much, but it would be sweet if it would work like this.
I'm personally 100% satisfied with everything at this point with the inclusion of the leader key, this will just make it 105% :)

squarology, 2016-05-28:
Isn't this possible already? I don't have my keyboard yet (*sighs*--soon, very soon) but I imagine you could program like you would a space cadet shift and just replace the bracket with KC_LEAD and the shift with a layer toggle.

Something like LT(Hotkey_Layer, KC_LEAD)

Hold that key and hit your hotkey layer's e and email@address_com is sent and if you hit that key and e then ErgoDox is sent.

Just throwin' something out there. If I'm wrong please let me know. Like I said, I haven't gotten my keyboard yet; I'm pretty sure it arrives Monday.

eltang, 2016-05-28:
@squarology No, `KC_LEAD` is a 16-bit keycode, not a regular 8-bit one—it won't fit into the `LT()` macro.

squarology, 2016-05-28:
good to know @eltang btw where do you find that information? I didn't know there were those kinds of  limitations. My planned layout has a space shift key... dear god I hope the space bar is an 8-bit key.

BeatVids, 2016-05-28:
Plus, I tried it just to see what would happen. Results were pretty weird, so I'm certain it doesn't work, at least the way I did it up above.

eltang, 2016-05-28:
@squarology The keycode for `KC_LEAD` is in `keymap_common.h` and all the normal keycodes are in `keycode.h`. Don't worry, the space key uses an 8-bit keycode, as do all the other keys that can be found on traditional keyboards.

jackhumbert, 2016-05-28:
Right now, `LT()` and friends only work with 8bit keycodes because it's just a shortcut for the TMK ACTION, which doesn't support the fancier 16bit keycodes that are used elsewhere in QMK. 

This is a little bit of a tangent, but it is possible to call the ACTIONs in `process_action_user` by doing something like this:

     if (keycode == KC_LEAD) {
       action_t action;
       action.code = ACTION_DEFAULT_LAYER_SET(0);
       process_action(record, action);
       return false;
      }

We messed around with the tap vs hold stuff while implementing it (I implemented the chording stuff at the same time), but found that it was pretty hard to differentiate between them in practice, since it's pretty common to press the second key down before completely releasing the first in a sequence (eg `TAP KC_LEAD , a`). It's possible to watch for the key release of the leader key, but it got a little complicated and nuanced to think about/describe accurately, so we opted for the current implementation :)

BeatVids, 2016-05-29:
Ok thanks Jack and @eltang once again for the help.

I was certain you guys gave it a go since it was clearly stated in the doc that's it's not possible.
I hope maybe you can include it someday, but I yet have to play with the leader key still, I'll throw some more thoughts and ideas if I think of any :)

closed 2016-07-24 by ezuk