qmk/qmk_firmware #357: alps64 fails to build

dsvensson, 2016-05-28:
Subject sums it up. alps64 target fails to build. After updating Makefile to include quantum.mk, and adding TOP_DIR here and there - it comes a bit further, but ultimately fails because its configuration doesn't contain MATRIX_ROW_PINS and MATRIX_COL_PINS. Is this an old, now unsupported, target that should be removed?

OT: On a side note, I found this when looking for targets not containing MATRIX_COL/ROW_PINS as I'm trying to figure out how to add support for the pegasus hoof controller. The row configuration is straight forward enough to make a MATRIX_ROW_PINS of, but the column one I have no idea how to translate into MATRIX_COL_PINS:

https://github.com/BathroomEpiphanies/epiphanies_tmk_keyboard/blob/master/controllers/pegasus_hoof_20150108/matrix.c#L185

eltang, 2016-05-28:
The alps64 keyboard is not very active on QMK, and the code is a bit outdated. You might have better lucks starting from scratch by running by pretending your keyboards are hand wired and following [the hand wiring guide](https://github.com/jackhumbert/qmk_firmware/blob/master/HAND_WIRE.md). It doesn't look like either of these two keyboards you mentioned have anything special about their matrices that can't be accommodated by QMK's `matrix.c`.

dsvensson, 2016-05-28:
There was a comment here a couple of minutes ago that referred to the hand writing guide. I've read it, and it didn't help with MATRIX_COL_PINS. The code I link to above is what I assume tells me what the COL_PINS should be set to, but I don't get how to make the translation. Even looking at how the pins were defined in QMK at an older point in time doesn't seem to help:

https://github.com/jackhumbert/qmk_firmware/blob/d66aa0abf96b5e887250cf0a7fa5e575f18c5a91/quantum/config_common.h#L4

...as some of the case's in that switch is way out of bounds of those pin definitions. Should each part of the 17 MATRIX_COL_PINS array values be or'ed like this? 

    #define MATRIX_COL_PINS { D5 | D4 | D1 | D0, D6 | D5 | D4, .... }
                             -------col0-------|-----col1----|-colN-

jackhumbert, 2016-05-28:
Sorry, I deleted that because it was pushing things in the wrong direction. We recently pruned a lot of projects, and the alps64 survived that. I've just pushed changes that should allow it to compile - do you actually have a board to test it on?

dsvensson, 2016-05-28:
Nopes, just looked at it as it didn't have any MATRIX_COL_PINS defined and I hoped I learn something from that, but then I realized it didn't compile, and that it probably lacked COL_PINS due to being outdated and filed this ticket. So I'm back at square 0.

jackhumbert, 2016-05-28:
Haha, getting me all excited for nothing! Thanks for pointing it out.

closed 2016-05-28 by jackhumbert

dsvensson, 2016-05-28:
Any suggestions to my pin problems though? How to express that with the MATRIX_COL_PINS style qmk uses?

jackhumbert, 2016-05-28:
The matrix.c file you linked uses two multiplexers (74HC42s), which currently aren't supported in the quantum matrix stuff. You can just pull in that matrix file by declaring `CUSTOM_MATRIX = yes` like the Ergodox EZ does, and add `matrix.c` to your `SRC` (both in the Makefile).

The MATRIX_*_PINS definitions are C arrays, so fancier notation with the pipes won't/wouldn't work.

dsvensson, 2016-05-28:
Yep, wrt the pin defs I was referring to the older representation where they were still just numbers. But CUSTOM_MATRIX sounds exactly what I'm looking for. Thanks.

jackhumbert, 2016-05-28:
Ah, gotcha. That would have been a neat side effect of that implementation!