qmk/qmk_firmware #358: Ergodox EZ: lights flash repeatedly when connected to shut-down Windows computer

adiabatic, 2016-05-29:
All this testing was done with three machines:

- spreadthinly, an iMac12,2 dual-booting OS X and Windows 10 (my main machine)
- paddedontheinside, a machine running Windows 10
- bluehal, a ProLiant MicroServer running FreeBSD 

I've noticed that my Ergodox EZ will flash its three lights in quick succession (first red, then green, then blue in a sort of wave) at weird times. This multicolored flash, which I'll call "rainbowing", happens repeatedly, while the computer is off, if I shut my computer off directly from Windows 10. 

Ordinarily this would be harmlessly cute, but I sleep in the same room as the keyboard and I don't want to have to either reboot into OS X just to make the blinkenlights stop shining.

Here's what I've noticed after a bit of experimentation: 

- The keyboard rainbows repeatedly during the boot-up procedure on spreadthinly. The rainbowing stops before the pick-a-drive screen comes up when you hold alt.

- when shutting down in Windows, you see a rainbow once, even before the hard drive parks.

- I shut spreadthinly off from Windows and the keyboard started rainbowing. I then unplugged it, and it stopped rainbowing (no surprise there). I then plugged the keyboard into a shut-off Windows machine (paddedontheinside) and it started rainbowing again. Finally, I plugged the keyboard back into spreadthinly and it started to rainbow again.

- I shut spreadthinly off from OS X. I then plugged the keyboard into paddedontheinside while the latter was off and it rainbowed only once.

- While spreadthinly was in OS X, I unplugged the keyboard and plugged it into the turned-off Windows machine (paddedontheinside). The keyboard started rainbowing repeatedly. I then plugged the keyboard into bluehal and the rainbowing didn't start up again.

- If I unplug and replug my keyboard into paddedontheinside, it'll stop blinking and resume blinking. Same thing with spreadthinly.

---

Is there anything else I can do to have my keyboard not flash lights at me when I shut down straight from Windows in the evening?

jackhumbert, 2016-05-29:
Do you have your firmware on github? There are some Makefile options to mess around with, namely `SLEEP_LED_ENABLE`. This really sounds more like the keyboard power cycling, though. If you look in `ergodox_ez.c`, you can remove `ergodox_blink_all_leds();` from `matrix_init_kb`, and this should stop.

adiabatic, 2016-05-30:
> Do you have your firmware on github?

[Yup](https://github.com/adiabatic/qmk_firmware/tree/master/keyboard/ergodox_ez/keymaps/zweihander-osx)!

If I stick `SLEEP_LED_ENABLE = no` in `…/ergodox_ez/keymaps/zweihander-osx/makefile.mk`, will it get picked up automatically by the typical `make clean && make KEYMAP=zweihander-osx` invocation?

jackhumbert, 2016-05-30:
Yeap! :)

ezuk, 2016-06-01:
@adiabatic did it do the trick?

ezuk, 2016-06-01:
And thanks @jackhumbert!

adiabatic, 2016-06-01:
Yup, that did it. Thanks, @jackhumbert!

closed 2016-06-01 by adiabatic