qmk/qmk_firmware #363: can't get lead keys to work despite following the instructions on the readme

squarology, 2016-05-31:
Hey, guys, any help on this issue would be greatly appreciated. I put `#define LEADER_TIMEOUT 300` near the top of the colemak layout (just testing to see if I could get the lead key to work). Then I put `LEADER_DICTIONARY() {
    leading = false;
    leader_end();

    SEQ_ONE_KEY(KC_F) {
      register_code(KC_S);
      unregister_code(KC_S);
    }
    SEQ_TWO_KEYS(KC_A, KC_S) {
      register_code(KC_H);
      unregister_code(KC_H);
    }
    SEQ_THREE_KEYS(KC_A, KC_S, KC_D) {
      register_code(KC_LGUI);
      register_code(KC_S);
      unregister_code(KC_S);
      unregister_code(KC_LGUI);
    }`within `matrix_scan_user` and when I compile it I get the following errors: 

Compiling C: keymaps/default/keymap.c
keymaps/default/keymap.c:4:0: warning: "LEADER_TIMEOUT" redefined [enabled by default]
 #define LEADER_TIMEOUT 300
 ^
In file included from ./ergodox_ez.h:4:0,
                 from keymaps/default/keymap.c:1:
../../quantum/quantum.h:52:0: note: this is the location of the previous definition
   #define LEADER_TIMEOUT 200
 ^
In file included from ./ergodox_ez.h:4:0,
                 from keymaps/default/keymap.c:1:
keymaps/default/keymap.c: In function 'matrix_scan_user':
../../quantum/quantum.h:59:34: error: 'leading' undeclared (first use in this function)
  #define LEADER_DICTIONARY() if (leading && timer_elapsed(leader_time) > LEADER_TIMEOUT)
                                  ^
keymaps/default/keymap.c:164:5: note: in expansion of macro 'LEADER_DICTIONARY'
     LEADER_DICTIONARY() {
     ^
../../quantum/quantum.h:59:34: note: each undeclared identifier is reported only once for each function it appears in
  #define LEADER_DICTIONARY() if (leading && timer_elapsed(leader_time) > LEADER_TIMEOUT)
                                  ^
keymaps/default/keymap.c:164:5: note: in expansion of macro 'LEADER_DICTIONARY'
     LEADER_DICTIONARY() {
     ^
keymaps/default/keymap.c:164:5: warning: implicit declaration of function 'timer_elapsed' [-Wimplicit-function-declaration]
In file included from ./ergodox_ez.h:4:0,
                 from keymaps/default/keymap.c:1:
../../quantum/quantum.h:59:59: error: 'leader_time' undeclared (first use in this function)
  #define LEADER_DICTIONARY() if (leading && timer_elapsed(leader_time) > LEADER_TIMEOUT)
                                                           ^
keymaps/default/keymap.c:164:5: note: in expansion of macro 'LEADER_DICTIONARY'
     LEADER_DICTIONARY() {
     ^
In file included from ./ergodox_ez.h:4:0,
                 from keymaps/default/keymap.c:1:
../../quantum/quantum.h:54:31: error: 'leader_sequence' undeclared (first use in this function)
  #define SEQ_ONE_KEY(key) if (leader_sequence[0] == (key) && leader_sequence[1] == 0 && leader_sequence[2] == 0)
                               ^
keymaps/default/keymap.c:168:6: note: in expansion of macro 'SEQ_ONE_KEY'
      SEQ_ONE_KEY(KC_F) {
      ^
make[1]: *** [obj_ergodox_ez/keymaps/default/keymap.o] Error 1
make: *** [all] Error 2

I've compiled other layouts including my own but I can't seem to get lead keys to work. Any help would be greatly appreciated. I feel like I'm missing something obvious. Thanks in advance!

eltang, 2016-05-31:
Move the definition of `LEADER_TIMEOUT` to your `config.h` file and put `#include "timer.h"` and `LEADER_EXTERNS()` at the top of your keymap file. See if you can compile after that.

squarology, 2016-05-31:
thanks brother! it worked. I did what you said including putting a semi-colon after `LEADER_EXTERNS()` and it worked sure enough. I really appreciate it, Eric.

closed 2016-05-31 by squarology

eltang, 2016-05-31:
Awesome! Sorry that I didn't realize that `LEADER_EXTERNS` needed a semicolon.