qmk/qmk_firmware #369: KC_LSPO / KC_RSPC needs to be configurable

dsvensson, 2016-05-31:
Currently it's assumed that ( and ) are located on Shift + 9 resp Shift + 0. This is not the case when using ISO layout host-side so it would be a good idea to expose this for customization in the keymap C-file.

Do I understand correctly that the firmware says what kind country code or something it should be identified as to the OS? If this is the case, then just specifying country code in the layout would be enough?

jackhumbert, 2016-05-31:
That's fair. I don't think the firmware/keyboard communicates that to the OS - we were messing with 104/105 key keyboards some on #312, but I'm not sure that's the best way to go.

How about making 4 #defines defaulted to shift, shift, 9 and 0 and sticking them [here](https://github.com/jackhumbert/qmk_firmware/blob/master/quantum/quantum.c#L438)? People can then override them in their keymaps/config files.

dsvensson, 2016-05-31:
Sounds like an ok fallback plan. I'll pop in the original controller in my keyboard and see if I can identify some difference. I don't recall it changing keymap when replugging before - but I could be wrong.

ezuk, 2016-06-01:
@dsvensson would you like to contribute a PR with the proposed solution?

dsvensson, 2016-06-02:
Sure, just verified that there's no interesting difference from lsusb POV between the original controller and the pegasus hoof one. I'll look into it.

ezuk, 2016-06-03:
Sounds good!

jackhumbert, 2016-06-18:
Addressed in b70248f - the keys are customisable, but the shifts are not for now. ISO layouts look to only use those.

closed 2016-06-18 by jackhumbert

dsvensson, 2016-06-19:
Cool, except the comment is misleading or?

> you can define those in your keymap like this

That has to be done via -DLSPO_KEY=KC_9 parameters in the Makefile, as the keymap is a C file and quantum.c includes nothing that's in any way defined in the keymap, or am I missing something?

I kind of got stuck thinking about other ways around this, as there seems to be a bit of a bias against ISO all over the source code. Perhaps it would be possible to have something more general in the keymap Makefile that declares it to be ISO or ANSI and then all the affected macros are redefined accordingly.

jackhumbert, 2016-06-19:
Sorry, yeah, those should be included in your `config.h` file, not the keymap. 

Yeah, the eng_us (not really ANSI or ISO) bias goes a bit deeper than that - the keycode names are based off those names as well, and I think the actual HID protocol is the same. It's the standard to assume everything's eng_us, and customise things from there. There are lots of different language options in the `quantum/keymap_extras/` folder.