qmk/qmk_firmware #371: Using massdrop configurator

piotr-dobrogost, 2016-06-01:
At https://ergodox-ez.com/pages/our-firmware we read that:

> Massdrop, the innovative startup behind the ErgoDox kits, has an online graphical configuration utility which you can use to easily create your own **layout**. Layouts you create using this configurator are fully compatible with the ErgoDox EZ, as it’s simply an ErgoDox.

However on configurator's page there's this information:

> This tool generates the Teensy board **firmware** .hex file for controlling the ErgoDox

I guess layout can't be separated from firmware and both go into single .hex file. In this case the question is what firmware does configurator put? I think it's not qmk…

ezuk, 2016-06-01:
Yeah, in the documentation above, layout == firmware. It's the same thing for all intents and purposes, and that document was written for non-coders. 

The Massdrop configurator outputs... something. It's certainly not QMK, and we can't tell you exactly what it is, because it's entirely closed source (yay!).  So no, you won't be getting the latest QMK features etc.

We have a graphical configurator of our own in the works, as well as [this wonderful](https://github.com/sboesebeck/ErgodoxLayoutGenerator) project from Stephen Boesback, which is available for download at https://www.boesebeck.name/elg-srv.zip.

I am currently on the hunt for a good freelance Rails dev who believes in the cause and wants to complete our graphical configurator for a reasonable budget. If you know of anyone, do share the word :)

piotr-dobrogost, 2016-06-01:
>The Massdrop configurator outputs... something. It's certainly not QMK, and we can't tell you exactly what it is, because it's entirely closed source (yay!). So no, you won't be getting the latest QMK features etc.

Well, in that case I'm not sure if directing people to use it without such information is the best thing to do.

ezuk, 2016-06-01:
Until we have our own graphical configurator, it certainly is. Some people are comfortable using QMK. Others need just basic functionality but want to move some keys around. I do appreciate your feedback but based on my interactions with thousands of customers, this is a good link to have there right now.

piotr-dobrogost, 2016-06-01:
OK, thank you for information.

closed 2016-06-01 by piotr-dobrogost