qmk/qmk_firmware #375: ErgoDox EZ preventing machine from sleeping

djl, 2016-06-03:
So this is a bit of weird issue. I'm not sure if this is specific to the QMK firmware, the keyboard itself or something else entirely.

When I have my ErgoDox EZ connected to my computer (running El Capitan 10.11.5) and put it to sleep, the computer does sleep but then immediately wakes up again. If I disconnect the EZ before putting the computer sleep then everything works as expected.

Checking the syslog after the machine wakes shows this line:

    6/2/16 19:45:11.000 kernel[0]: Wake reason: UHC2

Usually there's another line or two with more specific details about the wake reason (e.g the specific USB port or Bluetooth controller) but I don't have any of those.

Not sure if it matters but I don't experience this when using my ErgoDox Infinity.

Has anyone else run into this?

ezuk, 2016-06-03:
Heya! I hadn't run into this myself. A couple of questions:

1) Do you use the stock ErgoDox EZ firmware? Or a custom build? If the latter, could you link to your layout?

2) Does it happen when you flash the EZ with firmware from the Massdrop Configurator?

djl, 2016-06-03:
> 1) Do you use the stock ErgoDox EZ firmware? Or a custom build? If the latter, could you link to your layout?

This happens with both the default firmware and a custom layout.

I just noticed one difference between the default and my [custom layout](https://gist.github.com/djl/25716af8e1352bbbc05d394e4a4d85d2). When sleeping the machine with the default layout active, right before the machine wakes, the LEDs light up in the same manner as if the keyboard was booting.

> 2) Does it happen when you flash the EZ with firmware from the Massdrop Configurator?

I can't test this right now as it looks like the configurator is having issues - it won't let me download the .hex or .zip files. If anyone has a hex file from the configurator I'll flash it and see how it goes.

ezuk, 2016-06-03:
[simple_firmware.zip](https://github.com/jackhumbert/qmk_firmware/files/297508/simple_firmware.zip)

here you go -- this is the stock Massdrop firmware. Can you try and tell me what happens?

As for the LEDs flashing - my take on that is that it's starting to animate the "computer sleeping" pattern (LED sleep) but then wakes up, so you only see it once.

djl, 2016-06-03:
Ah! I don't have the issue when using the Massdrop firmware.

ezuk, 2016-06-03:
Okay great, so we know it's (A) firmware, not hardware, and (B) QMK specific. Now it's a matter of figuring out what exactly is going on.

@jackhumbert ever hear of something like this? Any thoughts?

jackhumbert, 2016-06-04:
Do you have a bigger chunk of the kernel log? There might be some additional information there that could be helpful.

jackhumbert, 2016-06-04:
Could you try setting `SLEEP_LED_ENABLE  = no` in your Makefile?

djl, 2016-06-05:
> Do you have a bigger chunk of the kernel log? There might be some additional information there that could be helpful.

Upon a reboot and a fresh firmware build I'm now seeing some USB related lines:

    6/5/16 15:48:55.000 kernel[0]: 095902.945797 PRT5@fd500000: AppleUSBEHCIPort::interruptOccurred: portSC 0x00301005
    6/5/16 15:48:55.000 kernel[0]: 095903.415289 PRT5@fd500000: AppleUSBHostPort::setPowerStateGated: kPowerStateOn -> kPowerStateSleep
    6/5/16 15:48:55.000 kernel[0]: 095903.415298 PRT5@fd500000: AppleUSBEHCIPort::suspend: 
    6/5/16 15:48:52.000 kernel[0]: Wake reason: EHCI
    6/5/16 15:48:55.000 kernel[0]: Previous sleep cause: 5
    6/5/16 15:48:55.000 kernel[0]: 095903.776073 PRT5@fd500000: AppleUSBHostPort::setPowerStateGated: kPowerStateSleep -> kPowerStateOn
    6/5/16 15:48:55.000 kernel[0]: 095903.776086 PRT5@fd500000: AppleUSBEHCIPort::resume: 
    6/5/16 15:48:55.000 kernel[0]: 095903.809906 PRT5@fd500000: AppleUSBHostPort::setPowerFloorGated: waiting for power level >= 2
    6/5/16 15:48:55.000 kernel[0]: 095903.809923 PRT5@fd500000: AppleUSBEHCIPort::interruptOccurred: portSC 0x00301005
    6/5/16 15:48:55.000 kernel[0]: full wake promotion (reason 1) 56 ms

That's from directly after the machine wakes up. Other than that I don't have anything unusual in my log.

> Could you try setting `SLEEP_LED_ENABLE = no` in your Makefile?

I have that set already. It doesn't help. Here's all the options I've tried so far:

    BOOTMAGIC_ENABLE = no
    COMMAND_ENABLE = no
    SLEEP_LED_ENABLE = no
    UNICODE_ENABLE = no

jackhumbert, 2016-06-05:
I think the EHCx code is from a usb device - I'm not sure what the UHCx is, but I'd imagine it's the same. 

That's interesting though - the wake reason is outputted 3 seconds before it's suspended. Is that accurate? Maybe that's caused by some logging thing though.

Just as a sanity check, you're `make clean`ing before making, right?

djl, 2016-06-05:
> That's interesting though - the wake reason is outputted 3 seconds before it's suspended. Is that accurate? Maybe that's caused by some logging thing though.

I think that's normal. It takes about 3 seconds from sleeping to when the machine wakes up again. Maybe the time is cached?

> Just as a sanity check, you're make cleaning before making, right?

Right.

jackhumbert, 2016-06-05:
Yeah, I'm not sure. [Here](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/common/avr/suspend.c#L109)'s one place in the code to mess around with.

The sleep LED stuff gets called [here](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/protocol/lufa/lufa.c#L229) - I kinda assumed that's where it was triggering the wake-up, but if you have that disabled, I'm not sure what would be causing it.

djl, 2016-06-05:
Alright, now we're getting somewhere :)

If I change `suspend_wakeup_condition` to immediately return `false` then the problem goes away. Now to find out why `matrix_key_count` in `egodox.h` isn't returning the expected value...

jackhumbert, 2016-06-05:
Ahh nice! It might have to do with the handling of the IO expander (the mcp23018). `matrix_power_up` isn't defined in the EZ's matrix.c file - maybe this needs to exist and have the `mcp23018_status = init_mcp23018();` (or other code) in it.

ezuk, 2016-06-07:
Is this something we want to tweak for the default?

djl, 2016-06-07:
Defining `matrix_power_up` and having it call `matrix_init` seems to fix the issue but I'm not familiar enough with the code to know if this is correct or whether it will cause any other problems.

jackhumbert, 2016-06-07:
Cool - thanks for confirming! I'll mess around with this in a bit :)

ezuk, 2016-06-07:
awesome, thanks @jackhumbert and @djl!

djl, 2016-06-09:
I may have spoken a little too soon. Calling `matrix_init` does fix the problem it seems to cause a new problem. Upon sleep the LEDs constantly cycle on and off in the same pattern as the boot pattern.

As I haven't actually seen what the LED sleep pattern looks like yet I'm not sure if this is it and the `SLEEP_LED_ENABLE` option is being ignored or if this is just the boot pattern being repeated over and over.

jackhumbert, 2016-06-09:
If you put this in the power up function, it should avoid the start-up led blinking that gets called from the `matrix_init_kb` function at the bottom of `matrix_init`:

```c
    // initialize row and col

    mcp23018_status = init_mcp23018();

    unselect_rows();
    init_cols();

    // initialize matrix state: all keys off
    for (uint8_t i=0; i < MATRIX_ROWS; i++) {
        matrix[i] = 0;
        matrix_debouncing[i] = 0;
    }

#ifdef DEBUG_MATRIX_SCAN_RATE
    matrix_timer = timer_read32();
    matrix_scan_count = 0;
#endif
```

It's just the `matrix_init` without that last line.

djl, 2016-06-09:
That seems super obvious in hindsight 😐 

It looks like this is now working perfectly. Thanks!

jackhumbert, 2016-06-18:
Added `matrix_power_up` to the ez's matrix file (and quantum's as well, even if it wasn't an issue there) at de326d0.

closed 2016-06-18 by jackhumbert