qmk/qmk_firmware #378: [Question] Is there a way of making a Fn key for OS X?

jtbandes, 2016-06-04:
The default firmware has keys for both ⌫ and ⌦. Having both of these is nice, but I don't need separate keys for them. On Apple keyboards there is a "fn" key, and pressing fn+⌫ is the same as ⌦. The fn key also affects the behavior of F1–F12, making them "pure" function keys instead of activating special features like brightness adjustment.

But I can't figure out how to add a fn key to my keymap. I didn't see any "modifier"-like constant defined for it, and `xev` doesn't show any events when it's pressed/released. However, the OS X "Keyboard Viewer" is able to show when the fn key is pressed and released, and the "EventViewer" included in [Karabiner](https://pqrs.org/osx/karabiner/) shows a FlagsChanged event with code 0x3f for the fn key.

Does anyone know if there's a way to do this, or should I just "fake it" using a custom layer that has the keys I want?

jtbandes, 2016-06-04:
I also just noticed another issue that might be somewhat related. On Apple keyboards, holding shift while pressing the volume up/down keys will disable the auditory feedback for the volume change. But for some reason, on the ErgoDox, holding shift while pressing KC_VOLU or KC_VOLD doesn't have the same effect.

jackhumbert, 2016-06-04:
Unfortunately the Fn key is limited to the Apple keyboards, and keyboards that directly emulate apple keyboards - this isn't something that's currently possible with the firmware. It'd be easiest to try to emulate the functionality with additional layers.

I'm not sure why the shift wouldn't modify the volume sounds, but my guess is it's related to the above. I don't know of a workaround for that :/

jtbandes, 2016-06-04:
Holding shift seems to work now...either I was confused or something mysterious has changed. Anyway, thanks for the response. I was hopeful that whatever Karabiner was able to detect would be something the keyboard could produce, but perhaps not. (Or maybe someone cleverer than me could figure it out by intercepting events from an Apple USB keyboard and trying to emulate those...)

ezuk, 2016-06-07:
Looks like this is as resolved as this is going to get! :) Closing it for now. Thanks @jackhumbert and @jtbandes

closed 2016-06-07 by ezuk