qmk/qmk_firmware #383: Color output during make

jackhumbert, 2016-06-05:
Are there any good reasons for not doing this? It could make things a bit more informative.

skullydazed, 2016-06-05:
As long as there's both a way to turn it off and it's smart enough to only colorize output when stdout is a tty, I don't see any reasons for not doing it.

(I don't think it helps much, personally, but lots of people seem to benefit from color.)

jackhumbert, 2016-06-06:
Added this at a785b00 for messing around with - really only does the warnings/errors, but is useful to see what's wrong. Haven't added the conditional to it yet.

ezuk, 2016-06-07:
Oooh, colors! 🌈

jackhumbert, 2016-06-07:
Here's a preview of the make reformatting:
[image]

ezuk, 2016-06-07:
Super nice. Very clear. Makes me want to fix all the warnings :)

jackhumbert, 2016-06-11:
Merged in from #395!

closed 2016-06-11 by jackhumbert