qmk/qmk_firmware #385: Moving core Makefile and make commands to the layout folder

jackhumbert, 2016-06-06:
Mentioned in #382, we're looking at moving things to the `keymaps/<layout>` folder in each project. This will do a couple of things:

1. Allow layout-specific build options to only be defined once (inside the layout's makefile)
2. Get rid of the fancier keymap-detecting code in the keyboard's makefile
3. Generate the .hex files directly in the layout folder without confusion of where things went
4. Keep the fancier makefile stuff in the keyboard's makefile, only needing to `include ../../Makefile` at the bottom of the layout's makefile.

Currently a lot of files/folders get created in the make directory - it'd be nice if this could be avoided by creating something like a `.build` (open to suggestions) folder in the layout folder, where everything but .hex would be placed. 

We'll need to echo something like "Please build in the keymaps/layout folder instead of here" or possibly build with the "default" keymap when `make` is run in the keyboard directory. I suppose we could still support the whole `keymap=layout` notation as well, nullifying 2 above. This would still make Makefile required for all layouts, even if it's just including the keyboard's.

I'd also like a standard name to exist (for all projects besides the EZ) for the .hex files - something like `firmware.hex` or `qmk.hex` (open to suggestions), so that the filenames in each layout folder will be more standardised.

eltang, 2016-06-06:
This will require users to provide their own Makefiles when adding new layouts, correct?

Also, I was reminded of an idea I had a while ago by this. Right now, the configuration of builds takes place in both the Makefiles and the `config.h` files. Would it be good to move these options into one place?

jackhumbert, 2016-06-06:
Yes, but it'll be recommended to copy it from the default/basic layout, just like it is now.

The config.h will stay in the keyboard folder, but a config file can be placed in the layout folder, where it will override/add to the parent.

eltang, 2016-06-06:
Well, I think that giving every keymap a `config.h` file might make a little more sense than giving every keymap a Makefile. It might be easier to have one Makefile per keyboard like now for ease of maintenance. Options such as `BACKLIGHT_ENABLE` could all be moved to the `config.h` files. All that would be needed to be done is add "include guards" to all the source files whose compilation is currently controlled by the Makefile.

jackhumbert, 2016-06-06:
It would be easier, but it's not possible. You need Makefile-level access to compile files.

jackhumbert, 2016-06-06:
The changes to `tmk_core/rule.mk` [here](https://github.com/jackhumbert/qmk_firmware/commit/d9b45e39118e6593d235e1453f084bc0f3a77990) reflect the `.build` containment mentioned here. Once created, the `.hex` file gets copied from `.build/*.hex` to the target folder

jackhumbert, 2016-06-06:
Updates at @7f85436:

* cleans before each build
* make from root with keyboard=keyboard, keymap=keymap
* make from keyboard/keyboard with keymap=keymap
* make from keymaps/keymap
* only implemented on planck

There's also better containment in `.build`. Clean also removes `.build` folders from each level. Because the paths are shared in VPATH, this is necessary.

jackhumbert, 2016-06-11:
Merged in from #395!

closed 2016-06-11 by jackhumbert