qmk/qmk_firmware #391: LSFT() keys such as KC_ASTR and KC_LPRN put the keyboard in a bad state if shift is held

ec1oud, 2016-06-09:
I'm using Ergodox EZ.  I have four such keys on my layout: KC_ASTR,KC_AMPR,KC_LPRN,KC_RPRN.  If I actually hold the shift key when I press one of those keys, I get into a weird state where most typing is impossible (except that the U key sends =, for some reason) and I need to reset the keyboard (re-plug it).

What I would like instead is to be able to assign other keys/macros to the shifted state for those keys.  The ultimate solution might be to treat the shift keys as a layer switch instead of sending shift, and then define what the shifted key will be on that layer.  Is such a thing possible?

eltang, 2016-06-09:
Most of what you want is already possible!

Change the `IS_COMMAND()` definition to a different condition or to `0` if you don't use the feature.

You can make a modifier key pop you into another layer every time it's pressed by making it `ACTION_LAYER_MODS(layer, mods)`. I'm not sure I understand your description of `KC_STRN`.

ezuk, 2016-06-16:
@ec1oud did this help?

sergei-dyshel, 2016-07-10:
I'm experiencing this problem. I have a layer (#1) which maps number keys to their shifted state (!, @, # etc.). When I'm pressing those keys alone, everything works perfectly. But as soon as I press Shift+number (while still being in that layer), keyboard start behaving strangely and I need to replug it. 

Any ideas?

jackhumbert, 2016-07-10:
It sounds like your keyboard's entering command mode, and changing the layer (shift + shift + number does this). You can disable command mode with `COMMAND_ENABLE = no` in your `Makefile`.

sergei-dyshel, 2016-07-10:
It was exactly that! Thanks, Jack!

ec1oud, 2016-07-13:
COMMAND_ENABLE = no in the Makefile: that works, thanks!

I didn't get around to trying the other suggestions yet.

sergei-dyshel, 2016-07-14:
Closing :smile:

closed 2016-07-14 by jackhumbert