qmk/qmk_firmware #392: [Enhancement] Leader key: Capitalizing the first letter by pressing shift first.

BeatVids, 2016-06-09:
Hi everyone, just had another idea regarding the leader key.

Let's say I have a combination to say "thank you"

    SEQ_ONE_KEY(KC_T)                { SEND_STRING("thank you"); }

What if I wanted to start off a sentence with "Thank you"? The macro won't be as effective because the initial capitalized T will be lower case. So my thoughts were to modify the leader function to not count shift keys for the leader_sequence_size, but keep in mind they have been pressed.

This is totally doable already:

    SEQ_ONE_KEY(KC_T)                { SEND_STRING("thank you QMK"); }
    SEQ_TWO_KEYS(KC_LSFT, KC_T)      { SEND_STRING("Thank you QMK"); }
    SEQ_TWO_KEYS(KC_RSFT, KC_T)      { SEND_STRING("Thank you QMK"); }

But this requires triple the length of the leader dictionary. With the modification, all that will suffice is:

    LEADER_KEY_SHIFT_MOD = true
    SEQ_ONE_KEY(KC_T)                { SEND_STRING("thank you QMK"); }

`KC_LEAD + KC_RSFT + KC_T` > "Thank you QMK"
`KC_LEAD + KC_T` > "thank you QMK"

Just throwing another idea your way :smile:

jackhumbert, 2016-06-11:
I think the implementation you have there is probably the easiest :) it would be cool to watch for *a* shift instead of the left or right one, though.

It'd be cool if you could do `SEQ_TWO_KEYS(KC_LSFT | KC_RSFT, KC_T)`, but the conditional won't allow that unfortunately. If we switched it to a switch/case, some cool stuff would be possible, like this (albeit completely different):

    SEQ_ONE_KEY(KC_A ... KC_F)

Which would accept any key from A to F. Do you see any use for that? The downside is that we're limited to ranges that exist within the keycodes (no jumping around). If we added another function, we might be able to get away with it, like:

    SEQ_ONE_KEY(SEQ_OR(KC_A, KC_F))

Which would work for A or F. Any use for that? These wouldn't be adding a whole lot to the implementation, since they'd just be c macros (only increase size if used).

BeatVids, 2016-06-11:
Hi Jack!
Personally, I wouldn't any use for that at all, I'd like to have each letter have it's own function, thus providing many more macros. But it's nice to learn another trick :)

I made my suggestion because it's easy to print out a macro in all caps, if we just hold shift, and then press the leader key sequences:

> thank you
> THANK YOU

^ (Actually typed with the leader key :smile: )

Now the only thing that's missing is capitalizing the first letter if I wanted to start off the macro in a sentence.

> Thank you

But I must use the workaround for now for certain strings.

Thanks Jack!
Filling out applications is a complete breeze with the leader key, as you can imagine with the constant filling in of common info :laughing:

closed 2016-07-24 by ezuk