qmk/qmk_firmware #393: KC_LEAD and LED in ErogDoxEz

moonrumble, 2016-06-09:
is there an easy way to turn on the LED while KC_LEAD function is accepting input keys?

jackhumbert, 2016-06-09:
Sure! You have access to `ergodox_right_led_on(uint8_t x);` where `x` is 1, 2, or 3, so you could do something like this:

```c
void leader_start(void) {
  ergodox_right_led_on(1);
}
void leader_end(void) { 
  ergodox_right_led_off(1);
}
```

The `leader_start` is executed when the leader key is pressed, and the `leader_end` is executed when the action is executed.

moonrumble, 2016-06-09:
worked like a charm. Thanks

is send_string functionality is complete? some how send_string doesn't work on my ergodox but other function do work.

jackhumbert, 2016-06-09:
It is, but it's SEND_STRING - send_string accepts addresses.

moonrumble, 2016-06-09:
Pardon my ignorance, what is the difference?

jackhumbert, 2016-06-09:
[Here's the definition](https://github.com/jackhumbert/qmk_firmware/blob/master/quantum/quantum.h#L29) - ones a pointer, the other handles the direct strings. You'll want to use `SEND_STRING("Woo")` :)

moonrumble, 2016-06-10:
This is not exactly how I was thinking about it. 

Let's say leader time out is 1000. when I pressed leader key, I want an LED to start. that LED should stay on while I am pressing the other sequence. 

It should clear an LED once the time out is complete. 

Currently it just blinks and immediately clears. I wanted a clear indication that LEADER key is still active .

jackhumbert, 2016-06-10:
That should be how it will work - what's your timeout? It'll turn off as soon as the timer's up. If you could paste your code in here with ``` before and after, that might help.

moonrumble, 2016-06-10:
After I press KEY_LEAD and P, LED stops but it takes some times to print the string. Ideally once you press LEADER key, LED should be on, it should stay on until SEND_STRING is complete or time out happens. 

Should I move leader_end to the end of LEADER_DICTIONARY method?
```
LEADER_DICTIONARY() {
    	leading = false;
    	leader_end();

    	SEQ_ONE_KEY(KC_P) {
            SEND_STRING("Abcdefghijkl");
        }
  }
```

jackhumbert, 2016-06-10:
Yeah, that should help - it's possible that send string takes a while to output, though. The keyboard is sending the characters as fast as it can, but the OS might take a while to process everything.

moonrumble, 2016-06-10:
I believe it is slower then MACRO but I do some more testing. Thanks

closed 2016-06-10 by moonrumble