qmk/qmk_firmware #394: How do I make a chorded key?

squarology, 2016-06-10:
Hello, does anybody know of a good method to create a chorded key?

For example, right now I am using autohotkey to send a "z" when I hit "k" and "m". It works well but the only problem is that I can't hit control and then hit "k" and "m" at the same time and expect to get control+z, instead it just sends a "z". 

So I was wondering if I could program the keyboard to send a "z" when I hit two different keys. I was thinking about creating a chorded key layer. Do you think that's my best option? I'm not even sure if that'll work; I'll have to test it out.

jackhumbert, 2016-06-10:
Are you looking to prevent the k and m from firing when you press them together? You have access to `bool process_record_user(uint16_t keycode, keyrecord_t *record)`, where that sort of control can be written.

squarology, 2016-06-10:
thanks, Jack, I'm pretty sure that's the first step. Off the top of your head do you know where the document is that explains that?

jackhumbert, 2016-06-11:
It's kind of a diy thing ;) one approach would be to look for the `KC_K` and `KC_M` keycodes, and turn on a boolean for each when they're pressed - you can then have a block that watches those two booleans for both keys being pressed down, and output `KC_Z` when they are. The side effect of that is you're always going to output either a `k` or `m` before your `z` (you can't stop both without a delay of some sort, which would be difficult to do well).

squarology, 2016-06-11:
I appreciate it, Jack. And thanks for the advice to get me started.

closed 2016-06-11 by squarology