qmk/qmk_firmware #400: Record macros on the fly

jayeheffernan, 2016-06-13:
First of all thank you, I am loving many of the advanced features of this firmware, including macros and dual-action keys.  That said, the one killer feature that I feel is missing from my Ergodox experience is on the fly recording of macros.  That is, press a "macro-record" button, press a key to record the macro to, press a bunch of keys that should be recorded, press macro-record to stop recording, then press "macro-play" and the key you selected to play back the macro.

Macros recorded in this way would not necessarily have to persist across boot.

Implementing this would have to include some sort tracking of key-press history, which, if done well, could be used for other stuff too.  For example, I would find it helpful to be able to define a macro to delete a word.  It could check the history, and delete back until the first non-word character.

eltang, 2016-06-13:
I know for a fact that Option-Delete deletes a word in OS X and I believe Ctrl+Backspace does the same thing in Windows.

cbbrowne, 2016-06-13:
I think what we've got here are two separate features:

1.  **Interactive key binding**.  It would be interesting to have an interface to allow you to specify a key mapping without reprogramming.  I suspect this one's tough.

2. **Record/Replay**.  
In Emacs world, C-x ( starts recording keystrokes into a macro; C-x ) terminates, and C-x e invokes The Macro.  
In vim, qa starts recording keystrokes into register a, qb into register b, and so forth, terminated by another "q".  @a will replay register a, @b register b, and so forth.
I imagine that this is what people have in mind; press [something] to start collecting keystrokes into [somewhere], press [something] to terminate that, and a third thing to replay those keystrokes.  It's interesting, though if you're using vim/Emacs, I think you already have processes that will be easier to work with.

My concern with record/replay is that:
a) You need a place to stow the data; might chew up limited-and-difficult-to-measure device memory
b) How do you report what you've got?  (Nasty if one of the keys was Control-alt-Delete!!!)
c) Gotta have an encoding for representing a set of buffers/registers/whatever if there's support for multiple of these recordings.

edge226, 2016-06-24:
I know how to code C... I personally found programming the macros to be too cumbersome and slow.

I do plan on pushing this upstream at some point when I am happy with it the functionaltiy. I do not really intend to make this a fully automated tool however as a helper program it makes writing macros far simpler.

Of course this should work with any keyboard using this firmware but I coded it for my Ergodox EZ.

[Ergodox EZ Unofficial Macro Maker](https://gitlab.com/edge226/ergodoxez-macromaker), It is written in Bash. So Windows and Mac people will probably have to produce a VM or docker image in order to use it.

### Usage.
1. Load the library script with `. ./newmacro.sh` 
2. Create a single macro with `newmacro "String"` This will output macro info to the terminal.
3. nmm function accepts multiple strings and outputs to newmacro.txt

jackhumbert, 2016-06-24:
We have the `SEND_STRING("blah")` function now too - does this accomplish the same thing?

edge226, 2016-06-24:
@jackhumbert Not sure. How do you use SEND_STRING?

jackhumbert, 2016-06-24:
You can use it in macros like:

```c
if (record->event.pressed) {
    SEND_STRING("Hello!");
}
```

:)

edge226, 2016-06-24:
On basic examples this does the same thing. On complex cases including return statements it seems to fail if you simply add a return statement into the string.

Mine works with something like this:
```
newmacro "Here is some text.
Here is a second line.
 "
```

A space is required at the end of a newline in order to register the newline for some reason but It does work. Then you can just remove the resulting space that was added during this process. I plan on having the space removed automatically in the future.

Mine also purposely adds 2 W(255) before a return sequence. As I said this allows the macros to work in chat without sending messages too quickly.

jackhumbert, 2016-06-24:
It supports newlines as `\n` and tabs (`\t`) as well. I never experienced any issue with overloading the text input, even with longer strings, but a delay could be added to it. I think the current `W` implementation is a hard wait (locks up the keyboard), which is something that could change in the send_string's implementation.

edge226, 2016-06-24:
Okay awesome. I definitely think this needs to be documented.

edge226, 2016-06-24:
I needed a `break;` after using SEND_STRING() but thats no big deal. Just letting everyone know.

jackhumbert, 2016-06-24:
The break's required by the switch statement, not the send_string function :) 

This issue is more about recording the macros - feel free to make another issue about the send_string stuff and it's lack of documentation!

edge226, 2016-06-24:
Will do! Thanks for the great responses. I'll definitely let you know where I'm having issues because there have been a few places.

edge226, 2016-06-25:
When using SEND_STRING() twitch accepts the newlines correctly instead of overflowing the timer. :+1:

ezuk, 2016-07-24:
@jackhumbert ping? :) (Would like to close if that's not possible.)

skullydazed, 2017-06-27:
I'm not sure when we added them but https://docs.qmk.fm/dynamic_macros.html seems to cover this use case.

closed 2017-06-27 by skullydazed