qmk/qmk_firmware #402: config_user.h not being included after the makefile reorg

algernon, 2016-06-13:
I just updated my checkout to the latest master, and `make KEYBOARD=ergodox_ez KEYMAP=algernon` fails to compile, because `config_user.h` is not being included, thus `ONESHOT_TIMEOUT` does not get defined, and as such, `has_oneshot_mods_timed_out` does not get defined, and the build fails at the linking stage.

I have made a few feeble attempts at fixing the issue, but have not managed to resolve it yet.

(As a side note, `make VERBOSE=true` does not work either, which is a shame, it would be really useful for debugging)

algernon, 2016-06-13:
To make things a bit clearer, I can copy the default ErgoDox EZ `config.h` to `keymaps/algernon/config.h` and do some modifications, but I'd prefer to have only the overrides in there... `#include "../../config.h"` results in an error too, by the way.

jackhumbert, 2016-06-13:
Naming it `config.h`, putting `#include "../../config.h"`, and only including overrides in there is the new way forward - if you're getting redefinition errors, you can use `#undef` prior to your definitions.

algernon, 2016-06-13:
Gah, I'm an idiot. I was getting file not found errors when I tried the `#include "../../config.h"` way, but it turns out that error was due to me using symlinks: for development, I symlink a checkout of my keyboard layout repo into the qmk checkout, and that confused the hell out of gcc - and me. I switched to bind mounts, and all is well now.

Sorry for the noise!

closed 2016-06-13 by algernon

jackhumbert, 2016-06-13:
Ah, gotcha. Symlinks are awesome, but I haven't found a way to use them that gcc, git, and Github tolerate :)