qmk/qmk_firmware #404: Improving debouncing

eltang, 2016-06-13:
I've made some new debouncing code which I've been using it myself with no problems, but I'm not sure if anyone else will notice a difference. Basically, the current code currently waits for the state of all the keys in the matrix to be stable for 5 ms before triggering an action. My new code triggers an action immediately after it detects a change and then ignores the just the key which triggered the action for 5 ms afterwards. I feel like my keyboard is snappier now, but I'm not sure if that's just me. It'd be great if I could get some additional opinions. My branch is [here](https://github.com/eltang/qmk_firmware/tree/debouncing_fix). The commit before the current commit (which works also) got rid of the delay, and the current commit implemented the per-key debouncing. I'm not sure whether one, both, or none of these commits should become part of QMK.

@TerryMathews

eltang, 2016-06-14:
@TerryMathews I think I know what the problem is, lemme update the branch real quick. Let's move our discussion over there so it doesn't get deleted (I like to update commits to keep my history clean).

eltang, 2016-06-14:
@TerryMathews Try it now.

closed 2016-06-14 by jackhumbert

eltang, 2016-06-14:
@jackhumbert What's wrong? I thought you told me to move the discussion somewhere else.

jackhumbert, 2016-06-14:
This isn't the place to ping people when you've updated your code - use your own fork, create issues there.

eltang, 2016-06-14:
@jackhumbert Well, I was hoping to get a few more people to try the code out by opening an issue here; my repo doesn't see as much traffic.

jackhumbert, 2016-06-14:
That's a pretty annoying approach to development, and doesn't belong in the issues section here.