qmk/qmk_firmware #413: Undefinied references with BLUETOOTH_ENABLE

gotteric, 2016-06-16:
Switching `BLUETOOTH_ENABLE` to `yes` causes the following:

`undefined reference to 'bluefruit_serial_send'

jackhumbert, 2016-06-16:
Which keyboard have you enabled it on? I can't reproduce it with the Planck.

gotteric, 2016-06-16:
Planck. 

Not sure if this is relevant, but I was also having issues with switching `AUDIO_ENABLE` to `no`, saying undefined reference to `eeconfig_read_audio`

jackhumbert, 2016-06-16:
You're using the most up-to-date code, right? What OS are you on?

gotteric, 2016-06-16:
I cloned on the 10th. And I'm on Windows 10.

jackhumbert, 2016-06-16:
Ah, pull the updates if you can - there was a big commit on the 11th.

gotteric, 2016-06-16:
Ok, update pulled. Now it's breaking at the beginning of `make` with the default keymap:

`/usr/bin/sh: awk: command not found
../../tmk_core/rules.mk:612: recipe for target '.build/obj_planck_default/keyboard/planck/planck.o' failed
make[1]: *** [.build/obj_planck_default/keyboard/planck/planck.o] Error 127
../../tmk_core/rules.mk:400: recipe for target 'all' failed
make: *** [all] Error 2`

gotteric, 2016-06-16:
`awk` and `gawk` are coming up as undefined in cmd, but working in git-bash

jackhumbert, 2016-06-16:
I'm not very familiar with Windows systems, but awk is needed for the new build system. Are you able to compile with git-bash?

gotteric, 2016-06-16:
I was just trying that. Now gives:

`/usr/bin/sh: -c: line 0: syntax error near unexpected token `('
/usr/bin/sh: -c: line 0: `C:/Program Files (x86)/MHV AVR Tools/bin/make begin '
../../tmk_core/rules.mk:400: recipe for target 'all' failed
make: *** [all] Error 1`

jackhumbert, 2016-06-16:
Make has the worst errors. Does `make COLOR=false` provide anything different?

gotteric, 2016-06-16:
Nope, exactly the same

jackhumbert, 2016-06-16:
Let me prune the `awk`s into conditionals and see where that gets us. Does `make show_path` in `cmd` work alright?

gotteric, 2016-06-16:
Yeah

`VPATH=../../keyboard/planck/keymaps/default ../../keyboard/planck ../.. ../../tmk_core ../../quantum ../../quantum/keymap_extras ../../quantum/audio ../../tmk_core/protocol/lufa ../../tmk_core/protocol/lufa/LUFA-git ../../tmk_core/common`

`SRC=keyboard/planck/planck.c keyboard/planck/keymaps/default/keymap.c quantum/quantum.c quantum/keymap_common.c quantum/led.c quantum/matrix.c quantum/audio/audio.c quantum/audio/voices.c quantum/audio/luts.c lufa.c descriptor.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Common/HIDParser.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/ConfigDescriptors.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/DeviceStandardReq.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/Events.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/HostStandardReq.c protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/USBTask.c common/host.c common/keyboard.c common/action.c common/action_tapping.c common/action_macro.c common/action_layer.c common/action_util.c common/keymap.c common/print.c common/debug.c common/util.c common/avr/suspend.c common/avr/xprintf.S common/avr/timer.c common/avr/bootloader.c common/magic.c common/avr/eeconfig.c common/mousekey.c common/command.c common/backlight.c common/avr/eeconfig.c`

jackhumbert, 2016-06-16:
Do you have access to `cat` in `cmd`?

jackhumbert, 2016-06-16:
I've just committed 33e83aa that should work if you have `cat`.

gotteric, 2016-06-16:
`cat` is working in cmd and git-bash. Doesn't work. Aside from a shift in the line number where the error occurred, it's the same.

jackhumbert, 2016-06-16:
Can you paste the full error? This is with `cmd`?

gotteric, 2016-06-16:
Yeah, it is `cmd`

`/usr/bin/sh: awk: command not found
../../tmk_core/rules.mk:618: recipe for target '.build/obj_planck_default/keyboard/planck/planck.o' failed
make[1]: *** [.build/obj_planck_default/keyboard/planck/planck.o] Error 127
../../tmk_core/rules.mk:406: recipe for target 'all' failed
make: *** [all] Error 2`

jackhumbert, 2016-06-16:
Can you try doing this to `/tmk_core/rules.mk`? (just forcing it not to use `awk`)

```diff
diff --git a/tmk_core/rules.mk b/tmk_core/rules.mk
index f8a83df..d1a9780 100644
--- a/tmk_core/rules.mk
+++ b/tmk_core/rules.mk
@@ -89,11 +89,11 @@ ifeq ($(COLOR),true)
        BOLD=\033[1m
 endif
 
-ifeq ("$(awk /dev/null 2>&1)", "")
-       AWK=awk
-else
+# ifeq ("$(awk /dev/null 2>&1)", "")
+       # AWK=awk
+# else
        AWK=cat && test
-endif
+# endif
 
 OK_STRING=$(OK_COLOR)[OK]$(NO_COLOR)\n
 ERROR_STRING=$(ERROR_COLOR)[ERRORS]$(NO_COLOR)\n
```

gotteric, 2016-06-16:
That seemed to work. I was able to run `make` and run `make dfu`.

jackhumbert, 2016-06-16:
Awesome - thanks for testing! I'll refine the test line a little bit.

gotteric, 2016-06-16:
No problem! (`BLUETOOTH_ENABLE` is working now too :smile: )

jackhumbert, 2016-06-16:
This is working on my machine - would you mind testing it when you get the chance? I'm debating just removing `awk` as a whole, haha.

```c
ifneq ($(shell awk --version 2>/dev/null),)
	AWK=awk
else
	AWK=cat && test
endif
```

gotteric, 2016-06-16:
Works

gotteric, 2016-06-16:
Seems to be working correctly

jackhumbert, 2016-06-16:
Awesome, thanks again for testing. Fixed at 91b469d.

closed 2016-06-16 by jackhumbert