qmk/qmk_firmware #414: Change pin used for audio

mohitg11, 2016-06-17:
Currently the firmware is setup to use the C6 pin for connection to a buzzer.

I'd created a PCB that uses some other pin, and would appreciate if there was an option to define the pin to use in the keymap/config file.

In general moving forward, I think it is a good idea to not have the pin used for this functionality hard-coded.

jackhumbert, 2016-06-17:
Unfortunately it's hard-coded because of it's reliance on Timer 3 - other pins can be used, but all of the timer code would need to be changed. What other pins do you have available?

mohitg11, 2016-06-17:
I'm using it on Pin B4, which I believe is Timer 2.

jackhumbert, 2016-06-17:
Are you using the Atmega32u4? I don't think B4 isn't associated with a timer on that chip.

jackhumbert, 2016-06-17:
The [datasheet](http://www.atmel.com/images/atmel-7766-8-bit-avr-atmega16u4-32u4_datasheet.pdf) has it listed as `PB4 (PCINT4/ADC11)`.

mohitg11, 2016-06-17:
You're right, I messed up my schematic when shifting from atmega90usb1286 to atmega32u4.

Ah well, looks like a hotwire.

jackhumbert, 2016-06-17:
Ah, dang. D0 is available on Timer 0, but that's currently used by the core stuff. That's the other complication - all of the timers are used somewhere, so it's a puzzle trying to get them to not overlap.

closed 2016-06-18 by jackhumbert