qmk/qmk_firmware #418: The compiled.hex files could be travis artificats

fredizzimo, 2016-06-18:
It would be great if we wouldn't have to manually update the compiled.hex files, and instead let the continuous integration system deal with it.

This would also take care of the annoying thing of having every compiled.hex modified, when you build all the keyboards to test your changes, and have to manually handle those in git. I have also not found any documentation on when to actually include the files in a pull request and when to not.

I'm not sure exactly how it should be done, but I think it's through [github release support for travis-ci](https://docs.travis-ci.com/user/deployment/releases). This would probably also allow us to generate a binary package that could be downloaded, so that end users don't even have to clone the git repository in order to flash new layouts.

jackhumbert, 2016-06-18:
That would be awesome! I'll look more into this tonight. It'd be great if we could ignore the .hex files again. 

It would definitely be something after the initial commit though, since that's what triggers the CI stuff. It would kind of suck if it doubled our commits.

jackhumbert, 2016-06-19:
So I got this implemented on the subprojects branch at this commit: https://github.com/jackhumbert/qmk_firmware/commit/eb63261172f520892a2d4e5691ecb1ec8fcff241 with an amend commit. I'm going to mess around with [this](https://github.com/dmakhno/travis_after_all) now, so I can have it run once at the end, inside of 14+ times.

jackhumbert, 2016-06-19:
This also creates an endless loop on the travis.ci side.. something else to address.

jackhumbert, 2016-06-19:
Dropped the script to just `make all-keyboards`, and it commits all updated .hex files: https://github.com/jackhumbert/qmk_firmware/commit/da4b601c2ba9f3e9c22575ed79d697c0d5dff74f - still has the endless loop you have to cancel manually, though.

Diffs are going to be a little hard to read with all of the files being updated, though. I'm debating just dropping the compiled.hexs all together.

jackhumbert, 2016-06-19:
I'm starting to think this would make more sense as a separate branch that gets populated only by Travis. It could even be only .hex files with a folder structure like

```
/
  alps64/
    default.hex
  arrow_pad/
    default.hex
    arrow24.hex
    ...
  ...
```

jackhumbert, 2016-06-19:
So after talking with @ezuk a bit, I think we're gonna have travis make a $(branch)-automated-build branch automatically, and update the compiled.hex files there - from there we can make a PR and merge things in manually on a weekly/daily basis, depending on how often we make changes. Users are still welcome to update their .hex files with their keymaps though.

This has been implemented in the subprojects branch, and will get merged in sometime this week.

fredizzimo, 2016-06-19:
Did you check out the [Github releases](https://help.github.com/articles/about-releases/), which Travis CI [supports](https://docs.travis-ci.com/user/deployment/releases), and I mentioned in the first post?

That way, I think it should be possible to generate a package that could be downloaded, without creating new commits in GIT. This package could include just the hex files, and possibly also the dfu-programmer. 

I'm not exactly sure how it works in practice though, since I have just glanced over the documentation.

jackhumbert, 2016-06-19:
Yeah, we have releases right now, but not that often (I've only done one that marks the Planck/Preonic rev 4/1 PCB firmware). I think we'd want to avoid doing that many releases though (one after each commit).

We still have the .hex files ready to download in the `keymaps/*` folder, but I like the dfu-programmer idea. This is still just a link away in a readme though :) the same goes with the Teensy Loader (for the Ergodox EZ/Teensy boards). One thing that's missing from this is having all of the .hex files together, but I'm not entirely sure if that's really necessary - this alone would be neat to do as part of a release, if we ever get to the point that we're releasing QMK at different points down the road. 

Ultimately, we want the compiled.hex files to exist in the current state, and be semi-up-to-date, and have the `master-automated-build` branch where they're completely up-to-date, and able to be merged into `master` when necessary.

fredizzimo, 2016-06-19:
Yes, I think the branching idea is the way to go.

If releases were to be used they should be quite infrequent, and tested. But that would probably need special branches, with extra testing. So it's probably not a good idea, at least not at this stage of the project. In that case, those who wants the latest features could always compile everything themselves,

languitar, 2016-06-20:
I'd be in favor to just have the hex files somewhere downloadable as an archive and keep them out of the repo at all. This will eventually just bloat the size of the repo until github starts to complain and people have to wait ages to pull it.

algernon, 2016-06-20:
> This will eventually just bloat the size of the repo until github starts to complain and people have to wait ages to pull it.

It's already quite large when you are on a limited connection and want to do a fresh checkout.

jackhumbert, 2016-06-20:
We'll be keeping in the repo for now - it doesn't have a huge impact on the size of the repo (compared to the total size). Right now all the `compiled.hex` files are 5.9MB. This is the command I used to add them:

    find . -type f -name 'compiled.hex' -exec du -ch {} + | grep total$

Eventually there will be an external site that will host the .hex files, and provide a nicer way to view/browse them - this will be done in coordination with @ezuk.

fredizzimo, 2016-06-20:
II agree with  @languitar.

While all hex files right now are only 5.9MB in total size it will quickly add up. Since Git needs to store the whole history, and I doubt that Git will find good deltas, it means that it's 5.9MB extra for each commit. 100 commits, and we are already talking about a half GB.

The number of keyboards and keymaps will also grow in the future, so that will make the space requirements even bigger.

fredizzimo, 2016-06-20:
Travis also supports uploading to [Amazon S3](https://docs.travis-ci.com/user/uploading-artifacts/).

The S3 free tier is probably enough for the numer of users that we have. But it's only valid for 12 months, so after that it would cost a little. But if I'm reading the pricing right, that would be less than 1 dollar a month.

ezuk, 2016-06-20:
My main concern here is in terms of UX for users who are not coders. Right now, we don't yet offer a graphical configurator (though this is in the works). Until that happens, we just send people to browse through the keymaps and grab what they want to try. I would not want to complicate this process in any way -- it is already a lot for a non-technical person to do. So any change we make has to also account for this and make things easier for laypeople (or at least keep the difficulty on par).

languitar, 2016-06-21:
Shouldn't deploying the rendered `README.md` per keymap as well as the hex file to some browseable site make things even easier than browsing the github repo, which is full of other stuff.

ezuk, 2016-06-22:
That would be wonderful -- the devil's in the details. How do we automate this exactly, etc. If we can figure that out, I am all for it, it would be much nicer. A nav menu on the left, then the keymap's readme on the right with a nice big Download button for the Hex. If we can automate that for all keyboards in the repo, that would be beyond amazing.

algernon, 2016-06-22:
Travis + GitHub pages could solve it, I suppose. I can prep a proof of concept sometime next week, if there's interest in seeing one.

On every push to master, build all keyboards, prep the readmes, and arrange them into a reasonable web page, commit the results to the `gh-pages` branch, and push them back. Done. Automated, no conflicts with the master branch, and the results are hosted on github too.

ezuk, 2016-06-22:
I like it! Would love to see a proof of concept. This doesn't solve the repo size issue mentioned above, but it sure makes for a nicer UI for end users IMHO.

languitar, 2016-06-22:
But doesn't the gh-pages branch also count to the repo size?

algernon, 2016-06-22:
It does, but `git clone --single-branch` may help with that. The hex files are text, so even if git stores many copies, they compress very well.

When I start on the PoC, I'll check how the hex files affect the size of the repo, and how fast they grow if regenerated often.

jackhumbert, 2016-06-22:
@ezuk and I just chatted about this a little, and I think we're gonna try making a `qmk.fm` repo and push the compiled files plus all keyboard stuff there to be served up into a webpage/site. This will keep the compiled files out of the qmk_firmware repo history and branches, and help to keep things a little more separate!

ezuk, 2016-06-22:
what he said. :tada:

jackhumbert, 2016-06-23:
If anyone's interesting, you can check out the site at http://qmk.fm. [The ErgoDox's page](http://qmk.fm/keyboard/ergodox_ez/) is pretty cool, and @algernon's [keymap](http://qmk.fm/keyboard/ergodox_ez/keymaps/algernon/) looks nice :)

jackhumbert, 2016-06-23:
The `subprojects` branch is now pushing there, so all of the code is tied to that (right now), except for the main readme - that's pulled directly on generation.

languitar, 2016-06-23:
> If anyone's interested, you can check out the site at http://qmk.fm. The ErgoDox's page is pretty cool, and @algernon's keymap looks nice :)

404 :/

algernon, 2016-06-23:
[This](http://qmk.fm/keyboards/ergodox_ez/keymaps/algernon/) is the correct link, an **s** was missing from the `/keyboards/` part of the URL.

And damn, it does look nice!

jackhumbert, 2016-06-23:
Thanks :) oops - updated that!

algernon, 2016-06-23:
FYI, the links on the site are still wrong, and point to `/keyboard/` URLs.

ezuk, 2016-07-24:
@algernon I believe this is now fixed. Could you confirm? If so, please feel free to close.

algernon, 2016-07-24:
Yep, the links look fine! I can't close it, since I'm not the original submitter of the issue, so pinging @fredizzimo :)

fredizzimo, 2016-07-27:
I'm closing this.

 I haven't personally verified that the artifact .hex files work. But at least they are downloadable from http://qmk.fm/. They are also not included in the main repository anymore and doesn't appear when running the git status command.

closed 2016-07-27 by fredizzimo