qmk/qmk_firmware #421: KC_MEH and KC_HYPR do not work as expected

sboesebeck, 2016-06-20:
Hi,

I tried to use KC_MEH and KC_HYPR on some keys on my ErgodoxEZ and unfortunately the behavior was not like a usual modifier, it seemed like it was treated as a normal keypress. Symptom: you hold the key and in case of HYPER all modifiers are held.
But as soon as typing a key, the modfiers are released and the key press is send as standard keypress to the OS. I could create the expected behavior using `ALL_T(KC_NO)`

Is that a bug or a feature? ;-)

eltang, 2016-06-20:
That's a bug which I noticed a while ago. I'm pretty sure this patch will work.

diff --git a/tmk_core/common/action.c b/tmk_core/common/action.c
index 078a03c..be6dea2 100644
--- a/tmk_core/common/action.c
+++ b/tmk_core/common/action.c
@@ -74,7 +74,7 @@ bool process_record_quantum(keyrecord_t *record) {
     return true;
 }
 
-void process_record(keyrecord_t *record)
+void process_record(keyrecord_t *record) 
 {
     if (IS_NOEVENT(record->event)) { return; }
 
@@ -129,7 +129,7 @@ void process_action(keyrecord_t *record, action_t action)
                                                                 action.key.mods<<4;
                 if (event.pressed) {
                     if (mods) {
-                        if (IS_MOD(action.key.code) || !action.key.code) {
+                        if (IS_MOD(action.key.code)) {
                             // e.g. LSFT(KC_LGUI): we don't want the LSFT to be weak as it would make it useless.
                             // this also makes LSFT(KC_LGUI) behave exactly the same as LGUI(KC_LSFT)
                             add_mods(mods);
@@ -142,7 +142,7 @@ void process_action(keyrecord_t *record, action_t action)
                 } else {
                     unregister_code(action.key.code);
                     if (mods) {
-                        if (IS_MOD(action.key.code) || !action.key.code) {
+                        if (IS_MOD(action.key.code)) {
                             del_mods(mods);
                         } else {
                             del_weak_mods(mods);

sboesebeck, 2016-06-20:
cool thing, but the code was `reformatted` so I cannot use it easily. Will it be merged at some point?

jackhumbert, 2016-06-20:
Yeah, these keys have a pretty poor implementation right now because they were designed mainly to be used in macros with their wrapper counterparts. 

Are the modifiers released before the keycode gets sent? ie are the modifiers affecting the key correctly for you before being released automatically?

sboesebeck, 2016-06-20:
no, the modifiers are released before the keycode is sent. Meaning: you hit `HYPR-A` and you end up with only an `a`.

When taking a look at the events, you see that all modifiers are hold as long as you don't type another key. Then they are release before the other key is registered...

jackhumbert, 2016-06-20:
Ah, gotcha. The keys are really just this

```c
#define KC_HYPR HYPR(KC_NO)
#define KC_MEH  MEH(KC_NO)
```

and are likely being cancelled due to some appropriate mod clearing. 

There are some better ways to implement these - I'll try to take a look at them tonight.

eltang, 2016-06-20:
@sboesebeck What's the problem?

eltang, 2016-06-20:
This could be another potential solution.
```diff
diff --git a/quantum/keymap.h b/quantum/keymap.h
index 979ab2d..cf6bbd0 100644
--- a/quantum/keymap.h
+++ b/quantum/keymap.h
@@ -300,8 +300,8 @@ enum quantum_keycodes {
 #define ALL_T(kc) MT(MOD_LCTL | MOD_LSFT | MOD_LALT | MOD_LGUI, kc) // see http://brettterpstra.com/2012/12/08/a-useful-caps-lock-key/
 
 // Dedicated keycode versions for Hyper and Meh, if you want to use them as standalone keys rather than mod-tap
-#define KC_HYPR HYPR(KC_NO)
-#define KC_MEH  MEH(KC_NO)
+#define KC_HYPR MEH(KC_LGUI)
+#define KC_MEH LCTL(LSFT(KC_LALT))
 
 #ifdef UNICODE_ENABLE
     // For sending unicode codes.
```

ezuk, 2016-06-20:
@eltang 

> That's a bug which I noticed a while ago. Try this patch.

Did you open an issue when you noticed the bug? Is there a patch with a fix?

harob, 2016-08-16:
Note that AFAICT `ALL_T(KC_NO)` is not a perfect substitute for `KC_HYPR`, as the former appears to also have "one shot modifier" behavior -- i.e. if I tap the hyper key and then later tap "a", "hyper-a" will be sent. I would rather not have that behavior, so a working KC_HYPR would be very helpful.

Is there any other workaround to get KC_HYPR like behavior without the one-shot-mod behavior?

drashna, 2018-03-25:
Is this still occurring?

drashna, 2018-10-21:
I suspect that this has been long since fixed. And since that is the case, I'm closing this issue.

If it's not, feel free to re-open this issue.

closed 2018-10-21 by drashna