qmk/qmk_firmware #423: KC_LGUI and KC_RGUI not working on my ergodox EZ

Nixsm, 2016-06-20:
Hi,
using QMK firmware, my designated key does not output KC_LGUI nor KC_RGUI, it does output any other key(KC_A, KC_O and so on). Using the Massdrop .hex file the keyboard outputs LGUI normally.
Is this a known problem?

jackhumbert, 2016-06-20:
This might be caused by holding the the space + LGUI key when plugging the keyboard in - if you hold space + backspace, if should reset the eeprom, and restore the GUI keys.

Nixsm, 2016-06-20:
Whoa, that was fast.
Tyvm.

closed 2016-06-20 by Nixsm

alexives, 2016-06-28:
@jackhumbert I'm having the same issue, but I can't seem to get this fix to work. I also tried uploading the hex file in quantum/tools with no result... I can however use GUI_T(KC_NO) just fine... any thoughts?

jackhumbert, 2016-06-28:
Are you on an EZ as well? Bootmagic may have been turned off, and if the EEPROM reset doesn't work, you can temporarily map `MAGIC_UNNO_GUI` somewhere on your keymap to unset that setting.

alexives, 2016-06-28:
Perfect! Actually, I'm on an original dox. I guess that's probably why this didn't work. The `MAGIC_UNNO_GUI` key worked like a charm!

Thanks for the quick response!

douglascamata, 2016-12-02:
Is there an easy fix for this? I have the same issue but i would like to know if it is possible to fix this without using the command line.

jackhumbert, 2016-12-02:
What keyboard are you using? The EZ implemented this functionality into the default layout shortly after this. I believe it's the grave key on the symbol layer.

douglascamata, 2016-12-02:
I'm using an ergodox (vanilla, i think), bought from falbatech.pl. But I'm using the configurator from Ergodox EZ webpage. Maybe this is what I'm doing wrong. I think I should use the configurator from Massdrop, maybe?

douglascamata, 2016-12-02:
BTW, it works if I configure it as `Command + KEY` but don't actually fill this key.

pfac, 2019-07-11:
Since I was facing this issue today, I'll leave it here for reference. Following on the workaround mentioned by @douglascamata, using `LGUI(0)` instead of `KC_LGUI` fixed this for me. Works both pressing the key on its own, and holding it as a modifier for any other key.

Tested on Pop_OS! 18.10 (Ubuntu based) with no issues so far.

Hope this helps someone else.

Spongman, 2022-03-26:
Just wanted to add that, for me, hitting `Fn`+`Win` fixed the issue.