qmk/qmk_firmware #427: BL_INC and BL_DEC not working on Satan

denolfe, 2016-06-21:
I just recently flashed qmk onto my GH60 Satan board. Everything appears to be working correctly, with the exception of the BL_INC and BL_DEC commands. Basically, they turn the backlight on or off, without any levels. The `BL_TOGGLE` however works correctly.

Any ideas on what could be going wrong? Thanks.

jackhumbert, 2016-06-21:
The backlight for the satan is on or off right now - it'd be possible to implement backlight levels like the Planck has, though. Would you be interested in helping with this?

denolfe, 2016-06-21:
@jackhumbert. Yes, I can definitely help out.

jackhumbert, 2016-06-21:
Awesome! Can you try out qmk_firmware/satan-backlight-levels?

denolfe, 2016-06-21:
Yep, I'll load that this evening and report back.

denolfe, 2016-06-22:
`BL_INC`,`BL_DEC`, and `BL_STEP` all work correctly.

However, the caps lock LED does not adjust along with the rest of the LEDs. I'm not sure if this is intended or not.

I also noticed, that if you use `BL_STEP` all the way through, when going from Off to Level 1, it briefly flashes to max brightness.

All minor things. I'd be perfectly happy if it were merged in this state.

jackhumbert, 2016-06-22:
That's intended - unfortunately the caps lock LED pin doesn't have PWM control like the backlight does. 

Yeah, that's consistent with other boards - been meaning to look into that!

Thanks for testing! I'll get this merged :)

closed 2016-06-22 by jackhumbert

denolfe, 2016-06-22:
@jackhumbert Awesome, let me know when you get that LED breathing put in. I'd be happy to test!

jackhumbert, 2016-06-22:
Will do! I'm working on abstracting that and the main backlight code out now - I should have something for you to mess with later this week.

jackhumbert, 2016-06-26:
Btw, you can enable the breathing with `#define BACKLIGHT_BREATHING` in your `config.h`. The functions are currently enabled by default (or documented), but you can add them to your keymap! There are some examples [here](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/atomic/keymaps/pvc/keymap.c)!

denolfe, 2016-06-27:
@jackhumbert I wasn't sure of all the steps to get this to work. I added the `#define BACKLIGHT_BREATHING`, but I could not find an example of adding anything breathing related to a keymap. Do you have an example of that? From the link you gave, it looks like a macro that needs to be added? Thanks for the help!

jackhumbert, 2016-06-27:
Yeap - that's right! [Here's a better example](https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/planck/keymaps/experimental/keymap.c#L240) - this basically dims the light, then comes back on :)

Do you have your keymap on github? I can give you a diff to try if that's easier!

denolfe, 2016-06-27:
My keymap is [here](https://github.com/denolfe/qmk_firmware/tree/master/keyboards/satan/keymaps/denolfe). I tried emulating your second example without success.

jackhumbert, 2016-06-27:
Ah, you might need to pull the latest changes first :)

denolfe, 2016-06-27:
Hmm, no success. I can try this another time or wait until someone else puts it on their keymap. Thanks for the help!