qmk/qmk_firmware #437: MT key doesn't register as a modifier immediately after having been tapped.

squarology, 2016-06-23:
I have a `SFT_T(KC_SPC)` key and when I press it to produce a space and then immediately press it again to hold it down to shift a key, I get another space and the unshifted key. If it's not immediate then it works perfectly, but if it's immediate then the modifier key doesn't work.

Is there some value I can change in the source code to minimize the time it takes an MT key to go back to being normal?

piotr-dobrogost, 2016-06-23:
From https://github.com/jackhumbert/qmk_firmware/blob/master/doc/keymap.md#4-tapping :

> Tapping is to press and release a key quickly. Tapping speed is determined with setting of `TAPPING_TERM`, which can be defined in config.h, 200ms by default.

You might also be interested in `IGNORE_MOD_TAP_INTERRUPT define` – see #142

squarology, 2016-06-24:
That was totally it. Tapping_Term was my issue. Thanks, Piotr!

piotr-dobrogost, 2016-06-24:
Was I right that the reason was not quick subsequent press of `SFT_T(KC_SPC)` after it was tapped first but (too) quick press of a key while `SFT_T(KC_SPC)` was depressed?
What value did you set for `TAPPING_TERM`? One researcher came to 267ms as the _optimum_ value – https://geekhack.org/index.php?topic=41685.msg1105565#msg1105565

squarology, 2016-06-24:
No, that wasn't the issue. The issue was `SFT_T(KC_SPC)` was taking too long to become a modifier key. Indeed, I lowered it to 150ms. It was only slight but it makes a difference. The optimum value of 267ms seems good if I'm typing for enjoyment or if I'm using one hand. Maybe I'll go back, I don't know now; I'm just going to test this setting out for the time being.

piotr-dobrogost, 2016-06-24:
> No, that wasn't the issue. The issue was SFT_T(KC_SPC) was taking too long to become a modifier key. 

That's what I described as
> pressing a key when SFT_T(KC_SPC) was depressed not long enough?

squarology, 2016-06-24:
Oh, yeah, that's it. I got confused with the wording. That's exactly what was happening. For some reason I was thinking depressed = pressed. My bad.

piotr-dobrogost, 2016-06-24:
From [Why is the term “depressed” often used to describe a button which is pressed?](http://english.stackexchange.com/q/21067/5685):

> ... depressed works just fine for the state of a button being pushed in.

To be totally clear:
We have two buttons (A and B) and the following events
1. A-down, 2. A-up, 3. A-down 4. B-down (5.1. A-up 6.1. B-up or 5.2. B-up 6.2. A-up]

I took your statement
> when I press it to produce a space and then immediately press it again to hold it down to shift a key

to mean you are concerned with the time between events 2 and 3 whereas I think the behavior is due to the time between events 3 and 4.

squarology, 2016-06-24:
That's a very clear way of putting it. The problem is definitely between 2 and 3. 5.2 and and 6.2 are how things are happening and the behavior was not as I was hoping so I modified the tapping term and that made it so that the time between 2 and 3 was not as big as before. I assume that `TAPPING_TERM` does that in addition to setting how fast an MT-like key has to be tapped to send its tapped form.

When `TAPPING_TERM 200` this is what happens:

1. A-down, 2. A-up, (170ms) 3. A-down 4. B-down then A is sent (the tapped key) and B is sent the unmodified tapped B key. It doesn't even matter which I let go of first, those two keys will be sent after the B key is pressed. For example: 1. A-down, 2. A-up, (150ms) 3. A-down (5 seconds) 4. B-down then A and B are sent as tapped keys and the B is not modified.

When `TAPPING_TERM 150` this is what happens:

1. A-down, 2. A-up, (170ms) 3. A-down 4. B-down then A modifies B as it should and no tapped A is sent because it was used to modify B. It seems that lowering the `TAPPING_TERM` also reduces the time needed for the A-key to return to normal (as a modifier).

squarology, 2016-06-24:
Alright, so even though `TAPPING_TERM 150` fixed my original issue, it created another issue: space not registering and `SFT_T(KC_C)` not registering because my thumbs are too slow.

Is there a way to maintain `TAPPING_TERM 200` and cut down the time it takes for `SFT_T(KC_SPC)` to become a shift key again after I press space?

piotr-dobrogost, 2016-06-24:
Sorry for nitpicking but saying that `SFT_T(KC_C)` does not register doesn't make sense as this symbol denotes two different actions requiring different criteria to be performed. It would be cool if you could describe what sequence(s) you want working using `key-[up|down]` events and timings between subsequent ones.

> Is there a way to maintain TAPPING_TERM 200 and cut down the time it takes for SFT_T(KC_SPC) to become a shift key again after I press space?

To know for sure one has to inspect source code.
Having said that I'm sure @jackhumbert knows details and is just waiting to share his knowledge with us :)

squarology, 2016-06-24:
well, about the `SFT_T(KC_C)` I won't go into details except mention that my thumbs are not that fast and `TAPPING_TERM 150` is too fast for them which is why I'm searching for an alternative solution. Even tapping a key for a fraction longer than `TAPPING_TERM 150` results in the key not being sent. Obviously, the keyboard is registering them but I mean the key showing up on the screen.

Alright, so I have a thumb key that is mapped to be `SFT_T(KC_SPC)` and it's on my right thumb cluster. So let's say I'm typing, I finish a sentence and I end it with `KC_DOT`, immediately, I tap `SFT_T(KC_SPC)` and a `(KC_SPC)` is sent, immediately, I press the `SFT_T(KC_SPC)` and hold it down and then tap `(KC_T)` afterwards. Instead of getting a shifted `(KC_T)` I get `(KC_SPC)` and an unshifted `(KC_T)`.

Now I could wait after tapping the `SFT_T(KC_SPC)` which comes after the `KC_DOT` but it's not very natural especially now that I'm getting used to the layout. So `TAPPING_TERM 150` kind of fixes this as it lowers the wait time (but it doesn't entirely fix the issue because even then double spaces and unshifted letters occur if I'm typing fast) but it makes the thumb cluster hard to use since my thumbs are somewhat slow.

So the solution could be to make it so that the wait time (or buffer time) isn't dependent on `TAPPING_TERM`. I commented out `action_tapping.c` but I commented out the /* scan buffer for tapping */ section but it still didn't fix the issue (I didn't expect it to). In an ideal world the wait time or buffer time should be as little as possible unlike `TAPPING_TERM` which is bit hard to lower (at least for me).

piotr-dobrogost, 2016-07-05:
This feature is controlled also by `IGNORE_MOD_TAP_INTERRUPT` mentioned in my first post.
It's defined here https://github.com/jackhumbert/qmk_firmware/blob/eafaba6b532a2f013955599fc7aa5c599e638fd3/keyboards/ergodox_ez/config.h#L52:

`#define IGNORE_MOD_TAP_INTERRUPT // this makes it possible to do rolling combos (zx) with keys that convert to other keys on hold (z becomes ctrl when you hold it, and when this option isn't enabled, z rapidly followed by x actually sends Ctrl-x. That's bad.)`

Did you play wit it?

piotr-dobrogost, 2016-07-05:
>  I assume that TAPPING_TERM does that in addition to setting how fast an MT-like key has to be tapped to send its tapped form.

@jackhumbert 
Is the above true? When using `MT(mod, kc)` does tapping the key reset a counter or not? In other words does tapping the key has any effect on how subsequent events are processed? I would guess it should not and the state of firmware after tapping such a key is exactly the same as before tapping a key.

jackhumbert, 2016-07-05:
Sorry, I'm not sure at first glance - MT is just [a shortcut for the `ACTION_MODS_TAP_KEY` functionality from TMK](https://github.com/jackhumbert/qmk_firmware/blob/master/quantum/keymap_common.c#L122), and I'm not super familiar with it.

We're planning on rewriting a lot of these down the road, at which point I'll have a better understanding of how they work :)

squarology, 2016-07-07:
> Did you play wit it?

I remember playing with a few things but in the end nothing changed the behavior that was causing the issue. I did look into https://github.com/jackhumbert/qmk_firmware/issues/142 but I don't think I was able to effect any king of change. I'll have to ask Wez about what he did and if that still works.

drashna, 2018-10-21:
Was this still an issue?

drashna, 2019-03-29:
I think this was resolved by `TAPPING_FORCE_HOLD`, though, it does cause other issues. 
https://docs.qmk.fm/#/feature_advanced_keycodes?id=tapping-force-hold

closed 2019-03-29 by drashna