qmk/qmk_firmware #442: Oneshot active utilityfunctions

TD22057, 2016-06-25:
This is an enhancement request to add functions to control oneshot behavior and add an action macro ACTION_MODS_ONESHOT_TOGGLE() to call the toggle function.  I'm planning on implementing this feature and figured this issue could be used for any discussion about whether you want a pull request for this or not.

    oneshot_toggle()
    oneshot_enable()
    oneshot_disable()

If oneshot is disabled, then any key marked as oneshot behaves normally. When oneshot is enabled (the default), keys marked as oneshot work that way. It does not turn any modifier into a oneshot modifier - only mods marked that way in the action map will work as oneshots. This just allows that behavior to be turned off if desired.

This makes it easy to experiment in a keymap with oneshot modifiers. They can be all be declared as oneshot mods and another key can be designated to turn the oneshot behavior on and off. So deciding to use oneshots or not is a software switch controlled from the keyboard instead of requiring a change to the firmware.

I have a pull request in at TMK for this but it doesn't include oneshot layer support since that was never merged in there.  https://github.com/tmk/tmk_keyboard/pull/350

Depending one what you think, this could be done in a two-stage pull request.  Part 1) would implement the functions listed above so this capability exists.  Hopefully that's not very controversial.  Part 2) would implement an action macro to make it easy to call the toggle function and include in a keyboard map.  That's only a convenience and uses up bits in the action struct so that could be left off.  Obviously that requires anyone using this to do a bit more coding in their keymap to call the toggle functions.

jackhumbert, 2016-06-25:
Nice. I like this - I would merge it! We are trying to reimplement things in the new process_record format (only in QMK), though. If you'd like to do that for this, that would be awesome too!

TD22057, 2016-06-25:
Where can I find info on the process_record format?  I see a few calls to that function but they seem to overlap with the one shot code.

jackhumbert, 2016-06-25:
You can see the individual ones setup [here](https://github.com/jackhumbert/qmk_firmware/tree/quantum-keypress-process/quantum/process_keycode) - they're all called from `quantum.c`. I just started working toward this today though :)

TD22057, 2018-10-22:
Feel free to grab the PR from TMK and update it.  I haven't had time to do any keyboard work in quite awhile.

drashna, 2021-03-01:
Added.

PR #4198 (2021-03-01): Add ability to toggle One Shot functionality
## Description

This provides a way to disable the "one shot" functionality of one shot keys.  

This is useful primarily for one shot mods, as some programs have issues with them (I'm looking at you, remote desktop connection, on Windows).  Disabling one shot keys turns mods into regular mod keys, and layer keys int `MO` keys.  

~~This setting isn't persistent, but considering adding it to `keycode_config` since this has 6 bits left, and would take 1 bit from the remainder.~~

Keep status in eeprom, with keymap config. 

## Types of Changes

- [x] [... cut]
closed 2021-03-01 by drashna