qmk/qmk_firmware #443: Can't I map 'CTL_T' with 'RCTL'?

potatogim, 2016-06-25:
Hi, I tried to map CTL_T to RCTL with my ergodox ez.

so I have change quantum/keymap.h like below that.

`
302 // Additional keycode for mapping LXXX_T/RXXX_T
303 #define LCTL_T(kc) MT(MOD_LCTL, kc)
304 #define RCTL_T(kc) MT(MOD_RCTL, kc)
305 #define LSFT_T(kc) MT(MOD_LSFT, kc)
306 #define RSFT_T(kc) MT(MOD_RSFT, kc)
307 #define LALT_T(kc) MT(MOD_LALT, kc)
308 #define RALT_T(kc) MT(MOD_RALT, kc)
309 #define LGUI_T(kc) MT(MOD_LGUI, kc)
310 #define RGUI_T(kc) MT(MOD_RGUI, kc)
`

and change my right hand 'CTL_T' mapping with 'RCTL_T' but it still input LCTL.

Does anybody let me know how can I map right-ctrl with CTL_T?

jackhumbert, 2016-06-25:
Unfortunately only the mod type (not handedness) gets passed into that function right now - we'd need another bit to do that. This will likely change in the future, though!

If you don't mind me asking, how does the left control not work here? Do you have something binded to right control specifically?

You can set up an older-style mod-tap with `FN0`/fn_actions like this in your `keymap.c`:

```c
const uint16_t PROGMEM fn_actions[] = {
  [0] = ACTION_MODS_TAP_KEY(MOD_RCTL, <kc>);
}
```

potatogim, 2016-06-28:
I'm using right-ctrl key as HOST key in remmina :-)

On your advice, I will try to older-style mod-tap like that.

Thanks for your answer!

closed 2017-06-27 by skullydazed