qmk/qmk_firmware #445: Unable to compile: avr-gcc.exe: error: unrecognized command line option '-fdiagnostics-color'

denolfe, 2016-06-27:
I pulled the latest changes and I am receiving the following error when attempting to `make` the default Satan keymap:

```
/usr/bin/sh: cc: command not found
Compiling: ../../keyboards/satan/satan.c [ERRORS]

avr-gcc.exe: error: unrecognized command line option '-fdiagnostics-color'

../../tmk_core/rules.mk:564: recipe for target '.build/obj_satan_default/keyboards/satan/satan.o' failed
make[1]: *** [.build/obj_satan_default/keyboards/satan/satan.o] Error 1
../../tmk_core/rules.mk:366: recipe for target 'all' failed
make: *** [all] Error 2
```

After digging, it looks like an `else` was removed from `rules.mk` on the pr: 13bb6b4 on line [134](https://github.com/jackhumbert/qmk_firmware/blob/master/tmk_core/rules.mk#L134). Hope this helps!

jackhumbert, 2016-06-27:
You can disable this temporarily by doing `make COLOR=false` - it should be excluding it automatically, though. Thanks for reporting!

jackhumbert, 2016-06-27:
Hey, @NoahAndrews - if you get the chance, could you help me figure out what a good way to divert errors to null is on any system? The

```makefile
ifeq ("$(shell echo "int main(){}" | $(CC) -fdiagnostics-color -x c - -o /dev/null 2>&1)", "")
	CFLAGS+= -fdiagnostics-color
endif
```

isn't working on Windows like I thought it would.

NoahAndrews, 2016-06-28:
To anyone who finds this, use the MHV AVR Tools command line environment instead of the standard Windows one to make colors work correctly. It may also be called Inferno AVR Tools.

@jackhumbert: I think we can close this now

jackhumbert, 2016-06-28:
Sounds good - thanks!

closed 2016-06-28 by jackhumbert