qmk/qmk_firmware #450: Strange behaviour of Unicode function.

Fahrradflucht, 2016-06-27:
I tried to implement a third layer which enables me to write German umlauts (äöü) and ß, while staying in US key layout. I didn't find another solution to this then mapping the keys in that layer to the corresponding unicode, but the behavior I got was very strange. 

If I hit the key that I mapped to `UC(0x00FC)`, which should be `ü` according to [Wikipedia](https://en.wikipedia.org/wiki/Latin-1_Supplement_(Unicode_block)) I get the following: `ººƒç`

algernon, 2016-06-27:
What OS are you using? Different operating systems have different unicode input methods, and maybe the default of QMK (osx, I think?) is not suitable for you.

Fahrradflucht, 2016-06-27:
I use OSX 10.11.5.

> What OS are you using? Different operating systems have different unicode
input methods, and maybe the default of QMK (osx, I think?) is not suitable
for you.

GitHub](https://github.com/jackhumbert/qmk_firmware/issues/450#issuecomment-228756535),
or [mute the
thread](https://github.com/
/AMn9nRbtg3xF-QZwHdbNzd04dpyQulY5ks5qP9nMgaJpZM4I_IHh.gif)

algernon, 2016-06-27:
What happens when you press left alt, then, while held, 00FC, and release alt too? Does that input the correct character, or does ti do the `ººƒç` too?

jackhumbert, 2016-06-27:
Are you using the Unicode Hex Input? Keyboard > Input Sources > Other (at the bottom) > Unicode Hex Input.

Fahrradflucht, 2016-06-27:
No I didn't. I was using US layout. Now I works just fine.

Thanks.

closed 2016-06-27 by Fahrradflucht