qmk/qmk_firmware #454: Turn led on if ins is on

MarkPearl, 2016-06-28:
Hi, I'm sorry if this is a noob question. I'm new to this but loving being able to reprogram my ErgodoxEZ.

Is there a way to determine if a state is on for a key? I have found sample code for checking if a layer is on, but have no idea if I can check if a mode is on?

~~~
uint8_t layer = biton32(layer_state);
~~~

My scenario:
I'm using an ErgodoxEZ, I want one of the LED lights to go on when insert mode is on (KC_INS)

algernon, 2016-06-28:
I think the easiest to do this would me a macro, say, `M(1)`. Then:

```c
static bool ins_mode = false;

const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) {
  switch (id) {
    case 1:
      if (record->event.pressed) {
        ins_mode = (ins_mode) ? false : true;
        register_code (KC_INS);
        if (ins_mode)
          ergodox_right_led_1_on ();
        else
          ergodox_right_led_1_off ();
     } else {
       unregister_code (KC_INS);
     }
     break;
  }

  return MACRO_NONE;
}
```

Alternatively, you can do something similar with `process_user()`, in which case you'd use `KC_INS` in the keymap, and some code like this:

```c
static bool ins_mode = false;
bool process_record_user (uint16_t keycode, keyrecord_t *record) {
  switch(keycode) {
  case KC_INS:
      if (record->event.pressed) {
        ins_mode = (ins_mode) ? false : true;
        if (ins_mode)
          ergodox_right_led_1_on ();
        else
          ergodox_right_led_1_off ();
     }
     break;
  }
  return true;
}
```

The advantage of the former is that it only runs when you press the `M(1)` key, but is a bit longer. The second runs on every keypress, and places more load on the controller, but is a few lines shorter.

You could do without the `ins_mode` variable, most likely, but this is - in my opinion - cleaner. So unless you are running out of memory, it's fine.

MarkPearl, 2016-06-28:
@algernon thank you - that is awesome and makes a ton of sense.

closed 2016-06-28 by MarkPearl