qmk/qmk_firmware #455: Document the amazing new Tap Dance feature

ezuk, 2016-06-29:
So we now have @algernon's awesome Tap Dance feature merged in! 

I need to document the feature and show an example of how it's used.

squarology, 2016-07-07:
Hi, @ezuk, I'm not trying to rush you in any way, I'm just curious how much progress you've made with some documentation and an example or two. Thanks.

algernon, 2016-07-07:
The original pull request, #451 has some examples, simple ones. Those are a reasonable start until better docs are available.

ezuk, 2016-07-09:
@squarology I actually appreciate the ping. I've now simply taken @algernon's excellent documentation from the PR, credited him, and put it right in the official readme. Please let me know if there are any edits, tweaks, or examples you'd like to make.

squarology, 2016-07-20:
Is it possible to have a tap dance key do one thing when it's unshifted and another thing when it is shifted?

algernon, 2016-07-20:
@squarology yes, but not easily. You will have to use a custom function, with the `ACTION_TAP_DANCE_FN` helper, and you can check whether shift is set there, and register a keycode accordingly.

piotr-dobrogost, 2016-07-20:
Having double tap a key act like tapping the key with AltGr pressed could be a nice feature for those having to use AltGr to insert characters with diacritical marks or other special ones.

algernon, 2016-07-20:
@piotr-dobrogost working on something like that. I have [something](https://github.com/algernon/qmk_firmware/commit/ede6151aa82e4c73aefe24c0bc73827f7de4f094) that works sometimes, when one modifier is pressed, but it's not done yet.

The goal is to allow `ACTION_TAP_DANCE_DOUBLE(KC_A, RALT(KC_A))` to do the right thing, and register `RALT + A` when double tapped.

Converting QMK modifiers to TMK codes to pass along to `register_code` is not as trivial as I would have hoped.

moonrumble, 2016-07-21:
@algernon It is the right thing to do. I would expect `register_code` method to take all the combination that i can use directly keymaps directly.

algernon, 2016-09-08:
I think this can be closed now, there is adequate documentation, with examples, in the readme now.

ezuk, 2016-09-10:
Agreed! Thank you @algernon!

closed 2016-09-10 by ezuk