qmk/qmk_firmware #456: Question about repo housing of custom matrix keyboards

robbiegill, 2016-06-29:
firstly, thanks for expanding and maintaining qmk!

I'm using a hand-wired 5x13 matrix keyboard that currently lives in my fork of qmk (https://github.com/robbiegill/qmk_firmware/tree/rdg-custom/keyboards/fivexthirteen). 
qmk development moves quickly and rebasing on current qmk can be a bit of a pain. 

Are you interested in hosting less common matrix layouts in this repo so they can reap the benefits of the mass refactors without later manual intervention?

There are obviously pros/cons, but it would certainly be nice to be covered under the travis sanity checks :)

jackhumbert, 2016-06-29:
I think the `keyboards/` folder should stay easy to navigate, so I'm a little conflicted about merging in handwired projects, but I would like to be able to accommodate custom projects like that. I think at this point, restricting projects to commercial/kit boards is the best way to go, but I think something like a `handwired` or `custom` folder could work out well for those! 

We'll need to modify some of the Makefile stuff to accommodate a structure like that though - I'll see if I have time to mess around with that sometime this week :)

robbiegill, 2016-06-29:
Totally makes sense - looks like a lot of the changes are moving to something that will be easier to maintain in the long run + directory shuffling won't be happening all of the time, so hopefully staying on top will become easier over time.

I'll play around with the makefile and alternate paths and post back if I have something reasonable before you get to it.

ezuk, 2016-06-29:
+1 for `handwired` :)

algernon, 2016-06-29:
@jackhumbert regarding the folder structure, I have something similar that I didn't submit yet, but does something similar: https://github.com/algernon/qmk-ergodox-crazies

I could likely come up with something less hackish, but this already works :)

With that, one would be able to say: `make KEYBOARD=handwired VARIANT=fivexthirteen`

jackhumbert, 2016-06-29:
Nice :) I like that for standard keymaps - being able to do `make keymap=default/colemak` would be nice. For the handwired boards, I'd kind of like to treat them as top-level keyboards, without needing any make arguments.

Does the `-DUSER_KEYMAP` line work though?

Also, those variants are pretty good :)

algernon, 2016-06-29:
> For the handwired boards, I'd kind of like to treat them as top-level keyboards, without needing any make arguments.

Hmm, so you want something like `make KEYBOARD=handwired/fivexthirteen`?

As for the others: what would `KEYMAP=default/colemak` build? 

The `-DUSER_KEYMAP` thing worked the last time I checked, but that was a week or two ago. I really should make that repo into a PR... tomorrow, perhaps!

jackhumbert, 2016-06-29:
Yeah, but they would likely build from the `handwired/fivexthirteen` directory.

The `default/colemak` would build the colemak variant of the default keymap (`/keyboards/<keyboard>/keymaps/default/variants/colemak`, maybe?) - I'd like to shelve this idea for a while, though. I'm not sure it's really necessary to have variants right now - it hides a lot of the keymaps available.

jackhumbert, 2016-06-29:
I'm not sure yet if this will be tied into the subprojects implementation yet (it's the same general idea, without the code borrowing), but let's at least wait until that's merged in to start, or make a PR to subprojects instead.

jackhumbert, 2016-06-30:
I'm trying out support for this at ba6fb23 - you can see an example in `/keyboards/handwired/onekey`

jackhumbert, 2016-06-30:
This is apart of master now - feel free to add your project to the `handwired` directory, and we'll handle the bugs (if any) from new issues! :)

closed 2016-06-30 by jackhumbert