qmk/qmk_firmware #457: Compiler warnings and linker errors in Planck experimental

fredizzimo, 2016-06-29:
When I'm trying to compile the experimental keymap of the Planck keyboard I get the following output
```
-------- begin --------
avr-gcc.exe (AVR_8_bit_GNU_Toolchain_3.5.0_1662) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: ../../keyboards/planck/planck.c                            [32;01m[OK][0m
Compiling: ../../keyboards/planck/keymaps/experimental/keymap.c       [33;01m[WARNINGS][0m
 | 
 | ../../keyboards/planck/keymaps/experimental/keymap.c: In function 'action_get_macro':
 | ../../keyboards/planck/keymaps/experimental/keymap.c:227:17: warning: implicit declaration of function 'breathing_speed_set' [-Wimplicit-function-declaration]
 |                  breathing_speed_set(2);
 |                  ^
 | ../../keyboards/planck/keymaps/experimental/keymap.c:228:17: warning: implicit declaration of function 'breathing_pulse' [-Wimplicit-function-declaration]
 |                  breathing_pulse();
 |                  ^
 | 
Compiling: ../../quantum/quantum.c                                    [32;01m[OK][0m
Compiling: ../../quantum/keymap.c                                     [32;01m[OK][0m
Compiling: ../../quantum/keycode_config.c                             [32;01m[OK][0m
Compiling: ../../quantum/matrix.c                                     [32;01m[OK][0m
Compiling: ../../quantum/audio/audio.c                                [32;01m[OK][0m
Compiling: ../../quantum/audio/voices.c                               [32;01m[OK][0m
Compiling: ../../quantum/audio/luts.c                                 [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/lufa.c                        [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/descriptor.c                  [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Common/HIDParser.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/ConfigDescriptors.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/DeviceStandardReq.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/Events.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/HostStandardReq.c [32;01m[OK][0m
Compiling: ../../tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/USBTask.c [32;01m[OK][0m
Compiling: ../../tmk_core/common/host.c                               [32;01m[OK][0m
Compiling: ../../tmk_core/common/keyboard.c                           [32;01m[OK][0m
Compiling: ../../tmk_core/common/action.c                             [32;01m[OK][0m
Compiling: ../../tmk_core/common/action_tapping.c                     [32;01m[OK][0m
Compiling: ../../tmk_core/common/action_macro.c                       [32;01m[OK][0m
Compiling: ../../tmk_core/common/action_layer.c                       [32;01m[OK][0m
Compiling: ../../tmk_core/common/action_util.c                        [32;01m[OK][0m
Compiling: ../../tmk_core/common/print.c                              [32;01m[OK][0m
Compiling: ../../tmk_core/common/debug.c                              [32;01m[OK][0m
Compiling: ../../tmk_core/common/util.c                               [32;01m[OK][0m
Compiling: ../../tmk_core/common/avr/suspend.c                        [32;01m[OK][0m
Assembling: ../../tmk_core/common/avr/xprintf.S                       [32;01m[OK][0m
Compiling: ../../tmk_core/common/avr/timer.c                          [32;01m[OK][0m
Compiling: ../../tmk_core/common/avr/bootloader.c                     [32;01m[OK][0m
Compiling: ../../tmk_core/common/magic.c                              [32;01m[OK][0m
Compiling: ../../tmk_core/common/avr/eeconfig.c                       [32;01m[OK][0m
Compiling: ../../tmk_core/common/mousekey.c                           [32;01m[OK][0m
Compiling: ../../tmk_core/common/command.c                            [32;01m[OK][0m
Compiling: ../../tmk_core/common/backlight.c                          [32;01m[OK][0m
Linking: .build/planck_experimental.elf                               [31;01m[ERRORS][0m
 | 
 | .build/obj_planck_experimental/keyboards/planck/keymaps/experimental/keymap.o: In function `action_get_macro':
 | C:\Users\Fred\Documents\Programming\qmk_firmware\keyboards\planck/../../keyboards/planck/keymaps/experimental/keymap.c:240: undefined reference to `breathing_speed_set'
 | C:\Users\Fred\Documents\Programming\qmk_firmware\keyboards\planck/../../keyboards/planck/keymaps/experimental/keymap.c:241: undefined reference to `breathing_pulse'
 | collect2.exe: error: ld returned 1 exit status
 | 
```

Which makes sense sine, I can't see that the BACKLIGHT_BREATHING define is enabled anywhere, and the functions in question only exist when in that case. At first I thought that it was my ChibiOS changes that messed things up, since the Travis-CI build is passing, but I got the same when trying to compile the master branch as well.

There's still a chance that it's something in my local configuration, but I guess the easiest way to confirm that would be that someone else tries to compile it first.

I can see one difference from the Travis-CI build, my compiler version is 4.9.2, and the Travis-CI one is 4.8.2.

jackhumbert, 2016-06-29:
Oops, yeah that should be on master - it looks like [I stuck the `BACKLIGHT_BREATHING` in the `quantum-keypress-process` branch](https://github.com/jackhumbert/qmk_firmware/blob/quantum-keypress-process/keyboards/planck/keymaps/experimental/config.h) instead!

fredizzimo, 2016-06-29:
It still doesn't explain why Travis doesn't detect the error though.

jackhumbert, 2016-06-29:
I think it has to do with how we're handling errors in the build script - I believe `make all-keymaps` catches that.

fredizzimo, 2016-06-29:
Ah yes, it does indeed fail https://travis-ci.org/jackhumbert/qmk_firmware/jobs/140608399, but it's reported as a success.

jackhumbert, 2016-06-29:
Sorry, I think I got that backwards - `make all-keyboards` catches that, I think. There aren't any tests on travis that reflect that, though.

jackhumbert, 2016-06-30:
Is this still applicable after the recent merge-ins?

fredizzimo, 2016-06-30:
The latest version compiles fine, so I think you can close this.

Have you done anything to the travis problem? Otherwise we should probably open a new issue about that.

jackhumbert, 2016-06-30:
I did, but I also changed the command for the script - I think it's taken care of with the `all-keyboards`.

closed 2016-06-30 by jackhumbert

fredizzimo, 2016-06-30:
I think that change slows down the Travis builds though, since it can no longer run different keyboards in parallel through different jobs. But again, that's a separate issue.

jackhumbert, 2016-06-30:
Yeah, it does, but we need there to be a single test for the after_success to only one once.