qmk/qmk_firmware #466: Document Chibios/ARM stuff form #465

jackhumbert, 2016-07-01:
https://github.com/fredizzimo/infinity_ergodox/blob/master/README.md

fredizzimo, 2016-07-01:
There are also some more documentation in https://github.com/jackhumbert/qmk_firmware/tree/master/tmk_core/protocol/chibios. But that's mostly internal stuff, and some of it isn't relevant anymore.

For the required packages, I'm testing the different Vagrant configurations and updating the avr_setup.sh script now. So once that's done the correct packages should be added to the documentation. I also think the the documentation should primarily refer to running that script, instead of running the install commands manually.

fredizzimo, 2017-05-24:
I think the general build setup instructions and the keyboard specific ones have enough information now.

closed 2017-05-24 by fredizzimo