qmk/qmk_firmware #467: Magical Secret Way?

blithen, 2016-07-01:
"There's also a magical, secret way to "lock" a modifier by tapping it multiple times. If you want to learn more about that, open an issue. :)"

I NEED TO KNOW JACK IT'S KILLING ME PLEASE

jackhumbert, 2016-07-01:
Haha, this was a joke between @ezuk and me, but I can't remember what the context was :)

I think you can just tap an OSM key more than once to lock it? Isn't that related to TAPPING_TERM? This is a portion of the code I'm not super familiar with!

algernon, 2016-07-01:
Haha, this is something I just explained to someone, over on reddit. Here's the trick: make your modifiers `F(n)` keys, and set up `fn_actions`, so the action for them is `ACTION_MODS_ONESHOT`, something like [this](https://github.com/algernon/ergodox-layout/blob/fe7541a6d218d15c1c6d1faf7eaf692bbca087f8/keymap.c#L533-L540).

You will also need to add `#define ONESHOT_TAP_TOGGLE 2` to your `config.h`.

ezuk, 2016-07-01:
This feature should only work on moonlit winter nights.

algernon, 2016-07-01:
> This feature should only work on moonlit winter nights.

If I ever build a custom keyboard, I'll add some thermal sensors, a camera and a clock, and one of the features WILL require a moonlit winter night. Mind you, I can get halfway there even with an ErgoDox EZ... hmm. 

*algernon adds it to his crazy-things bucketlist*

iFreilicht, 2016-08-15:
Just tested `ACTION_MODS_ONESHOT` and that in itself works fine, but these don't change a thing:

```
#define ONESHOT_TAP_TOGGLE 2
#define ONESHOT_TIMEOUT 1
```

I can set the timeout to whatever I want, the key never times out, and tap toggling doesn't work either. Guess I'll have to put Capslock on a function layer for now, but it would be a nice feature to have working.

cquiroz, 2017-06-07:
I have an ergodox and an atreus 62 and this feature works fine on the ergodox but not on atreus. `fn_actions` clearly is being called but when I double tap I don't get Shift locked

Any idea what could be the difference?

shieldsd, 2017-06-07:
try:
```
#define ONESHOT_TAP_TOGGLE   2
#define ONESHOT_TIMEOUT           500
```
Works for me. The default number of taps is five which I think is confusing.

cquiroz, 2017-06-07:
i tried that already and it doesn't change it. My timeout is longer though at 3000. I'll enable debug to learn a bit more

drashna, 2018-10-21:
If this is still an issue, try using `OSM(MOD_*)` for the keycode, rather than a function.