qmk/qmk_firmware #473: ergodox ez not waking up after sleep

qubitstx, 2016-07-04:
Been having an issue with the Ergodox-EZ (with QMK) not waking up after the computer goes to sleep. Forced to unplug, wait ~10 seconds, re-plug keyboard to get a keyboard reboot (led sequence flash), after a few seconds, the keyboard responds. Can't do a quick plug/re-plug, nor does it work immediately. Doesn't seem to be a hardware setting--that same port works fine with other keyboards. Is this something intrinsic to the Teensy microcontroller?  (Happens on two different windows systems.)

jackhumbert, 2016-07-05:
Are you running the latest firmware, or the factory default? We made a change recently to the EZ's `matrix_power_up` that may have an effect on that.

qubitstx, 2016-07-07:
Built from a May 26 version of the firmware, not factory default.  Will have to give matrix_power_up a try.

ezuk, 2016-07-27:
@qubitstx closing for now, please reopen if this is still a problem and we'll sort it out.

mys721tx, 2020-01-06:
Per comment from @drashna in #5585, Ergodox EZ is still experiencing this issue as of 0.7.106. #7784 does not resolve the issue as Ergodox EZ is a AVR board.

jackhumbert, 2020-03-13:
It may be possible to add logic similar to what #7784 did to [lufa.c](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/lufa/lufa.c#L558) to fix the AVR boards affected by this.

jackhumbert, 2020-03-13:
I believe this is the same issue as #1149, so I'm going to close this one since the other contains a bit more information/tests about the situation.

closed 2020-03-13 by jackhumbert