qmk/qmk_firmware #479: Unit testing library

fredizzimo, 2016-07-05:
The serial link library that I have made for the Infinity Ergodox includes some unit tests that I would also like to include. 

Currently those unit tests are written using the [CGreen](https://github.com/cgreen-devs/cgreen) unit testing library. I thought it would be good after watching [this presentation](https://www.youtube.com/watch?v=Wmy6g-aVgZI). However my experience was quite bad, especially regarding the output when there are errors. It certainly gets it's job done, but it's not of the quality that I'm used to. 

I also understand that you want to add unit tests in the future, and I too want to add more unit tests, so I think we should have a good unit testing library included.

I'm used to [Google Test](https://github.com/google/googletest), which I think is about as good as you can get in c++/c. The reason why I didn't use it for my serial link library, was that it needs c++. However I don't think that's actually a problem for the unit tests. The tests can be written in c++, while the tested code is written in c. Furthermore, it's very unlikely that we want to run any tests on the actual hardware itself, so the additional memory requirements of using C++ shouldn't matter.

It has another advantage compared to CGreen as well, CGreen requires CMAKE, while the Google Test only needs compilation of a single file.

I have tried to search for other alternative C testing libraries, but I haven't really found anything, that is good enough for me. 

So should I convert my tests to Google Test and include that, or do you have any other opinions?

fredizzimo, 2016-07-05:
Check is very close to Cgreen, this is from the Cgreen documentation

> The main difference from this tool and other xUnit tools, such as "Check", is that test results are not stored. Instead they are streamed to the reporter psuedo-class, one that is easily overridden by the end user.
> 
> The other main extra feature is the support for writing mock callbacks. This includes generating sequences for return values or parameter expectations.

I'm using that mocking functionality quite extensively in the serial link tests, and also generally when writing unit tests, so if Check is chosen, we need some additional mocking library. [Mockitopp](https://github.com/tpounds/mockitopp), is one alternative, but again, that's C++. Google test also comes with GMock for mocking.

jackhumbert, 2016-07-05:
I'm unfamiliar with the options here and unit testing in general, but I like the sound of Google Test!

fredizzimo, 2016-07-05:
Google test is definitely sold, and I have been using it since it was released in 2008. At that time I also made my own mocking library, since nothing good existed for C/C++. It actually turned out to be very close to Google Mock which was released a bit later. But since then I have completely moved on to the Google Test/Google Mock combination.

So unless there's any objections, I will integrate it and convert the serial_link library to use that. 

Since I have a lot of unit test experience (I did some unit testing even before I started using Google Test, and most of my programming since, have been written in a test driven way), I can also add some first unit tests for the keyboard functionality itself. That just so that we at least have the possibility of unit testing things, as it's much easier to add tests when the framework is there in place. But that will probably happen after I have got the Infinity Ergodox working.

ezuk, 2016-07-27:
Sounds good! Closing this one for now as it seems like we're all good with Google Test. It'd be amazing to have an actual test suite for qmk.

closed 2016-07-27 by ezuk