qmk/qmk_firmware #481: Unifying the make commands to upload the firmware

fredizzimo, 2016-07-05:
Currently you have to run different make commands for different keyboards, in order to upload the firmware.

I think we have at least the following

1. dfu - which runs dfu-program
2. teensy - which runs teensy-loader-cli
3. program - which doesn't seem to be used anymore, but used to run avrdude
4. dfu-util - which runs dfu-util (this is new, and used for programming the Infinity Ergodox)

All of these are of course needed, but we could have the keyboard makefile specify which one to use, and have a single command to be run for all of them. We could even leave the original commands in place and add a new one for that common command.

jackhumbert, 2016-07-05:
I'm a fan of that! I'm working on the avrdude programming one right now, too.

The blanket one could be `make pgm`, and something like `PROGRAMMER = dfu` in the Makefile. It would have the same overriding capabilities (from command line and subprojects) via something like `make pgm=dfu`.

Renaming `program` to `avrdude` makes sense, I think

algernon, 2016-07-05:
You can even have all of these, as is, no `PROGRAMMER` needed, and have a `pgm` target depend on the appropriate program target. So the ErgoDox EZ `Makefile` would have something like this:

```make
pgm: teensy
```

And then have a `.PHONY: pgm` in the main makefile, for good measures.

Pros: minimal changes, and allows the real target to be as complex as need be.
Cons: all keyboard makefiles need to be modified (but that's the case anyway)

fredizzimo, 2016-07-05:
The suggestion @algernon, proposed is close to what I had in mind. That would allow you to write make pgm for the keyboard default programmer, and make dfu-util for example if you want to override the programmer.

All keyboards would not necessarily have to be modified, but in that case they would just not support the pgm target, you would have to use the specific one instead.

jackhumbert, 2016-07-05:
That's perfect! I'm not sure why I suggested the `make pgm=dfu` - it's obvious to just keep the `make dfu` functionality :)

closed 2016-07-27 by ezuk

algernon, 2016-07-27:
Mmmmm.. why was this closed? AFAIK, it's not done yet, or is it?