qmk/qmk_firmware #483: Restructure the quantum and tmk_core/common directories/files

jackhumbert, 2016-07-05:
There's really no need for these folders to be separate. Some rearranging of the functions and files into folders might be nice too.

jackhumbert, 2016-07-05:
@fredizzimo mentioned moving serial link, things like audio, and other qmk-specific libraries to their own root-level folder.

fredizzimo, 2016-07-06:
See the discussion regarding that in #482.

I don't really have any preference myself, other than, that I would like too see that the libraries we have control over are separated from the completely external ones, like ChibiOS.

One thing to note, that renaming the tmk_core folder will probably not completely stop us from cherry-picking TMK commits. We can use the subtree merge strategy for that ´git cherry pick -X subtree=our_new_tmk_core_folder` But I'm not sure if it's possible to add multiple subtree options, which would be needed if we split it into multiple folders. I also think this only works when cherry picking from the tmk_core repository, since the subtree option doesn't have any from=to syntax.

Unless someone has a direct answer to how exactly the subtree merge strategy works, and what limitations it has, I think it would be worthwhile to do a bit of experimentation. I don't think it would be very wise to completely disabling the option of cherry picking, or making it very hard.

ezuk, 2016-07-13:
Let's close this for now -- @jackhumbert?

fredizzimo, 2016-07-13:
In addition to closing, maybe some label could be added, so that you know that it's something that you might want to get back to.

For example "backlog".

jackhumbert, 2016-07-14:
I'd like to keep this open for now - hopefully I'll be able to get to it soon :)

If we closed it, I would likely forget about!

ezuk, 2016-07-27:
Labelled as `enhancement` for now :)

drashna, 2018-10-21:
>I'd like to keep this open for now - hopefully I'll be able to get to it soon :)

🤣

jackhumbert, 2020-03-13:
> I'd like to keep this open for now - hopefully I'll be able to get to it soon :)
> 
> If we closed it, I would likely forget about!

What an innocent thought! Going to close this as it's been a part of our larger goals in QMK for a while now.

closed 2020-03-13 by jackhumbert