qmk/qmk_firmware #492: Make RGB controls proper keys

TerryMathews, 2016-07-07:
The RGB code is currently being handled as a macro, where the macro pushes the action out. This is in contrast to the backlight code which has dedicated keycodes and doesn't have to go through the pretense of defining macros.

It appears that making the RGB controls into proper keystrokes would only require changes in quantum\keymap.c and keymap.h. I think I can tackle those changes, but before I do I wanted to check and see if you were adverse to fixing that for some reason, perhaps an issue I'm not aware of.

jackhumbert, 2016-07-07:
Yeap! That's all that needs to be changed. It should be `quantum/quantum.c` and `quantum/keymap.h` though. The prefix `RGB_` should be a good one to use. If you can think of 3 character codes (`RGB_XXX`) that should work well with what we have already.

TerryMathews, 2016-07-07:
How about:

RGB_TOG
RGB_MOD
RGB_HUI
RGB_HUD
RGB_SAI
RGB_SAD
RGB_VAI
RGB_VAD

?

If you think that looks good, I'll get a pull request ready this evening after I test the changes on my Phantom. Since all we're doing is moving the action and not really changing the code itself, I would think that simple operation would be sufficient testing.

jackhumbert, 2016-07-07:
Sounds good!

TerryMathews, 2016-07-07:
So, a wrinkle:

action.code seems to be expecting some sort of return value. All of the rgb functions are void functions, so it's throwing errors. Any thoughts on the best way to fix?

TerryMathews, 2016-07-07:
Compiling: ../../quantum/keymap_common.c
                    ←[31;01m[ERRORS]←[0m
 |
 | ../../quantum/keymap_common.c: In function 'action_for_key':
 | ../../quantum/keymap_common.c:143:16: error: void value not ignored as it oug
ht to be
 |     action.code = rgblight_toggle();
 |                 ^
 | ../../quantum/keymap_common.c:146:16: error: void value not ignored as it oug
ht to be
 |     action.code = rgblight_step();
 |                 ^
 | ../../quantum/keymap_common.c:149:16: error: void value not ignored as it oug
ht to be
 |     action.code = rgblight_increase_hue();
 |                 ^
 | ../../quantum/keymap_common.c:152:16: error: void value not ignored as it oug
ht to be
 |     action.code = rgblight_decrease_hue();
 |                 ^
 | ../../quantum/keymap_common.c:155:16: error: void value not ignored as it oug
ht to be
 |     action.code = rgblight_increase_sat();
 |                 ^
 | ../../quantum/keymap_common.c:158:16: error: void value not ignored as it oug
ht to be
 |     action.code = rgblight_decrease_sat();
 |                 ^
 | ../../quantum/keymap_common.c:161:16: error: void value not ignored as it oug
ht to be
 |     action.code = rgblight_increase_val();
 |                 ^
 | ../../quantum/keymap_common.c:164:16: error: void value not ignored as it oug
ht to be
 |     action.code = rgblight_decrease_val();
 |                 ^
 |
../../tmk_core/rules.mk:396: recipe for target '.build/obj_phantom_default/quant
um/keymap_common.o' failed

TerryMathews, 2016-07-08:
And the actual code:

`#ifdef RGBLIGHT_ENABLE
		case RGB_TOG:
			action.code = rgblight_toggle();
			break;
		case RGB_MOD:
			action.code = rgblight_step();
			break;
		case RGB_HUI:
			action.code = rgblight_increase_hue();
			break;
		case RGB_HUD:
			action.code = rgblight_decrease_hue();
			break;
		case RGB_SAI:
			action.code = rgblight_increase_sat();
			break;
		case RGB_SAD:
			action.code = rgblight_decrease_sat();
			break;
		case RGB_VAI:
			action.code = rgblight_increase_val();
			break;
		case RGB_VAD:
			action.code = rgblight_decrease_val();
			break;
	#endif`

jackhumbert, 2016-07-08:
Sorry, that's why it needs to go into `quantum.c` - the `rgblight_toggle()`, etc are just functions you can call directly - you don't need the `action.code` stuff.

You'll also want to in case them in a `if (record->event.pressed)` block.

TerryMathews, 2016-07-08:
Ah, gotcha. Missed that part of your message. Oops!

TerryMathews, 2016-07-08:
PR#493

closed 2016-07-08 by TerryMathews